JuliaRobotics/DistributedFactorGraphs.jl #315: [BUG] setTimestamp! on DFGVariable missing

dehann, 2020-02-19:
```julia
ERROR: LoadError: MethodError: no method matching setTimestamp!(::DFGVariable{Pose2}, ::DateTime)
Closest candidates are:
  setTimestamp!(!Matched::Union{DFGFactorSummary, DFGFactor}, ::DateTime) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/CommonAccessors.jl:54
```

Affie, 2020-02-19:
What is your use case? There is a `setTimestamp` (note without the `!`) that creates a new variable since timestamp is an immutable field.
I was planning to do something like if it is needed: 
```julia
setTimestamp!(dfg::AbstractDFG, lbl::Symbol, ts::DateTime) = 
              updateVariable!(dfg, setTimestamp(getVariable(lbl), ts))
```

Affie, 2020-02-19:
https://juliarobotics.org/DistributedFactorGraphs.jl/latest/func_ref/#DistributedFactorGraphs.setTimestamp-Tuple{DFGVariable,Dates.DateTime}

dehann, 2020-02-19:
Use case is I'm building the factor graph and then later adding the timestamp information.  This comes from an existing function in RoME.  

xref JuliaRobotics/RoME.jl#240

dehann, 2020-02-19:
Also, the `!` should remain since you are updating the `dfg` object, is that no so?

dehann, 2020-02-19:
I think this is a `verb` issue: is the difference not perhaps `set` vs. `update`.  The `!` is indicating that some memory is being updated (no matter how small).

Affie, 2020-02-20:
> Use case is I'm building the factor graph and then later adding the timestamp information

Ok That still fits in. If it was something you wanted to change often the we would have to reconsider the struct. 
I think we should add timestamp as an argument in IIF `addVariable` that defaults to `now()` to show the importance of defining it.

> `set` vs `update`

In my mind we have 3 API groups:
1. **Accessors:** ie. `set` is a setter for fields only in the form `setField[!](object, value)`
2. **CRUD**: `get`/`add`/`update`/`delete` is  in the form `__Noun(object, key, [value])`
3. **Sets:** `list`/`merge`(set union)/`remove`(set diff)/`empty` is SET operations (WIP)

- 1 Can have different behavior depending on the driver. You have to keep track of where the object came from if working with something like solver data and call `updateVariable` for example, if you want to change it in the graph. Its more for advanced end-users and packages such as IIF.

- 2 and 3. should work the same everywhere. It is the interfaces that drivers such as CGDFG should implement. So a setField!(dfg, :lbl, val) kind of call will internally call `get`/`update` in CGDFG.

dehann, 2020-02-20:
Lets make sure we talking about the same thing?
- `setTimestamp(::DFGVariable, ts::DataTime)`
- `setTimestamp!(::AbstractDFG, ::Symbol, ts::DataTime)`

dehann, 2020-02-20:
Should we not do an API expose' in the docs, specifically on this timestamp example -- e.g here:
https://juliarobotics.org/Caesar.jl/latest/concepts/interacting_fgs/

closed 2020-02-21 by Affie

dehann, 2020-07-20:
> 1 Can have different behavior depending on the driver. You have to keep track of where the object came from if working with something like solver data and call updateVariable for example, if you want to change it in the graph. Its more for advanced end-users and packages such as IIF.

It feels like this is a bit in conflict with crossing desire -- i.e. #264

Affie, 2020-07-20:
> It feels like this is a bit in conflict with crossing desire -- i.e. #264

That is how it is with all fields. For example, it's impossible to change any field in a variable without the dfg object in CGDF.
 
So the 'crossing desire' can only be implemented with functions of the form `fncName(dfg::AbstractDFG, ...)`. To try and make everything the same would just slow it down unnecessarily since you can't pass by reference then. 

> Looking back, I still think only updateTimestamp! should be allowed and setTimestamp(...)::DFGVariable should not be made available. It's a small and seemingly unimportant issue at the moment, but perhaps one day we will be back here.

`updateTimestamp!` does not fit with the API design, we only have 'update' with full CRUD. Fields only support getters and setters. You update a variable to set its timestamp.

We can make the timestamp mutable and have a setTimestamp! at the cost of performance.

PR #322 (2020-02-21): [feature] setTimestamp!(dfg, lbl, ts) 
close #315
commit 8acd907c (2020-02-21): feature setTimestamp!(dfg,lbl,ts) close #315 (#322)