JuliaRobotics/DistributedFactorGraphs.jl #331: Feature request: More flexibility with `solvable`

Affie, 2020-02-24:
Related to checking if it is not solvable #222

I have some factors and variables that are not solvable with `:parametric`, but solvable with `:default`. 
I want to use a common fg and distinguish between them using `solvable`.
Since > is used I can not have a factor that only works on one, eg:
`:f1p` -> only used with parametric
`:f1d` -> only used with default 

We can consider moving solvable to `solverdata`, but then its only applicable to variables.
Again, why is it important to have > used?
What was the intended use case of Int apposed to bool, am I using it wrong?

dehann, 2020-02-28:
I'd say the purpose of supersolving is numerical differences with subtle structural differences, and hence solvable is kind of a consequence of the structure not the particular numerics...  I don't know if that is true or not.

The idea of solvable::Int > is to allow some freedom rather than a single binary 0/1.  The I have used it in the past is with setdiff:
```julia
onesbutnottwos = setdiff(ls(fg, solvable=1), ls(fg, solvable=2))
```

Affie, 2020-03-04:
For me > reduces the flexibility.

> onesbutnottwos = setdiff(ls(fg, solvable=1), ls(fg, solvable=2))` 

is all ok if you just want to list it, but you can't really  use it in the solver.

> The other reason for > is out of necessity, since multiple use cases exist where I want to know both solvable=0 and solvable=1.

That's just all of them with no filter. So the fastest way is not to check at all in this case. As it was originally.

What use cases , say a graph has solvable set:
[0, 1, 2, 8, 9, 20] - I can't see it ever being extremely large.

I think we aggree that
- [x] `isSolvable == getSolvable > 0`

Other
- [ ] `getSolvable == 2`
- [ ] `getSolvable in [1,2]`
- [ ] `getSolvable > 2`
- [ ] `getSolvable < 2`

I cannot think that we will have that many options for solvable. That's why my Vector suggestion in #222.

dehann, 2020-07-06:
Think we should get on call and resolve that way.

Affie, 2020-07-07:
We can consider using an optional argument, something like `; compare_with = >`

GearsAD, 2020-07-07:
+1 for a call

Affie, 2024-10-22:
See https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/222#issuecomment-2425997979 for the solution.

PR #1097 (2024-10-22): solvableFilter on listVariables and getVariables for GraphsDFG
TODO: still need to be added to all relevant functions and tests.

close #222
close #331
closed 2024-10-22 by Affie