JuliaRobotics/DistributedFactorGraphs.jl #342: getSolverData does not work without DFG.getSolverData

dehann, 2020-02-29:
some name collision here:
[image]
[image]

dehann, 2020-02-29:
```julia
(v1.3) pkg> status
    Status `~/.julia/environments/v1.3/Project.toml`
  [c52e3926] Atom v0.12.7
  [159f3aea] Cairo v1.0.2
  [5ae59095] Colors v0.11.2
  [a81c6b42] Compose v0.8.1
  [b5cc3c7e] DistributedFactorGraphs v0.6.2 [`~/.julia/dev/DistributedFactorGraphs`]
  [31c24e10] Distributions v0.22.5
  [186bb1d3] Fontconfig v0.3.0
  [c91e804a] Gadfly v1.2.0
  [a2cc645c] GraphPlot v0.3.1
  [904591bb] IncrementalInference v0.9.0 [`~/.julia/dev/IncrementalInference`]
  [033835bb] JLD2 v0.1.12
  [e5e0dc1b] Juno v0.7.2
  [2472808a] KernelDensityEstimate v0.5.2
  [c43967c8] KernelDensityEstimatePlotting v0.1.6
  [d41bc354] NLSolversBase v7.6.1
  [429524aa] Optim v0.20.1
  [91a5bcdd] Plots v0.29.3
  [295af30f] Revise v2.5.3
  [91fb55c2] RoME v0.6.0 [`~/.julia/dev/RoME`]
  [238d586b] RoMEPlotting v0.1.8 [`~/.julia/dev/RoMEPlotting`]
  [276daf66] SpecialFunctions v0.10.0
  [37b6cedf] Traceur v0.3.0
  [136a8f8c] ValueShapes v0.6.3
  [ade2ca70] Dates 
  [8bb1440f] DelimitedFiles 
  [8ba89e20] Distributed 
  [37e2e46d] LinearAlgebra 
  [10745b16] Statistics 

julia> using Dates; now()
2020-02-29T17:35:27.645
```

dehann, 2020-02-29:
> i seem to be able to use them when i use master of IIF and DFG?
<sub>[View in Slack](https://caesarjl.slack.com/archives/CKGRFPZT8/p1583017383016600)</sub>

GearsAD, 2020-02-29:
Just a quick test but I don't see that with master DFG and master IIF:
[image]

```
using DistributedFactorGraphs
using IncrementalInference

dfg = LightDFG{SolverParams}()

v1 = addVariable!(dfg, :a, ContinuousScalar, labels = [:POSE])
addFactor!(dfg, [:a], Prior(Normal(0,1)))

getSolverData(getVariable(dfg, :a))
getSolverData(getFactor(dfg, :af1))

```

dehann, 2020-03-03:
weird, thanks

closed 2020-03-03 by dehann