JuliaRobotics/DistributedFactorGraphs.jl #362: tag v0.7.1

dehann, 2020-03-04:
@JuliaRegistrator register branch=release/v0.7

JuliaRegistrator, 2020-03-04:
Registration pull request created: [JuliaRegistries/General/10527](https://github.com/JuliaRegistries/General/pull/10527)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.1 -m "<description of version>" 734861526a0d3a86e98919333d3883ff497c9aa2
git push origin v0.7.1
```

dehann, 2020-03-04:
closed by JuliaRegistries/General#10527

closed 2020-03-04 by dehann