JuliaRobotics/DistributedFactorGraphs.jl #372: CloudGraphsDFG connectivity bug

Affie, 2020-03-31:
@GearsAD, it looks like `getNeighbors` returns the source node if it has PPE relationships. 

I added -[r:FACTORGRAPH]- as relationship type in #371, but don't know if its correct.

There may also be other similar bugs.

dehann, 2020-05-25:
Hi @GearsAD , just a reminder if you had any input perhaps -- in preparation for DFG v0.8.0?

GearsAD, 2020-05-25:
Hi, will check it out tomorrow :+1:

GearsAD, 2020-05-28:
Interesting yes that seems to find the PPE and relate it as one of the connections - this is because a PPE matches to `n`. I had a similar issue with the `neighborhood` functions. A similar fix would be to force the source (`n`)  to be either a :FACTOR or a :VARIABLE (match (n:testUserId:DefaultRobot:Session_6a5ffd:a)--(node) where (node:VARIABLE or node:FACTOR) and node.solvable >= 0 and (n:VARIABLE or n:FACTOR) return node). 

I checked out `getNeighborhood` as well, just to see, and it looks ok. Think this can be closed.

closed 2020-05-28 by Affie