JuliaRobotics/DistributedFactorGraphs.jl #391: simplify `addDataEntry!`

dehann, 2020-04-21:
## Tasks

- [ ] vote on API including standardized Noun(s) to use
- [ ] implement simplified function and background DataStore pointers
- [ ] Update Noun/keyword definitions on the wiki.
- [ ] open corresponding issue for getDataEntry!

## Description

Perhaps simplify the current:
```julia
    # Make a big data entry in the graph - use JSON2 to just write this (really really verbosely)
    element = GeneralBigDataEntry(fg, systemstate.cur_variable, :RADAR, mimeType="application/json")
    # # Set it in the store
    addBigData!(datastore, element, Vector{UInt8}(JSON2.write(msg)))
    # # Add the entry to the graph
    addBigDataEntry!(systemstate.cur_variable, element)
```

to this:
```julia
addDataElement!(dfg, :var|:fct, :RADAR, JSON2.write(msg), mimeType="application/json")
```

### Must Define Nouns and keywords

- `Entry`
- `Element`

Also
- `mimeType`

## Related
- JuliaRobotics/Caesar.jl#514

Affie, 2020-04-21:
I think it can be a convenience wrapper, but please leave the separate functionality of `addBigDataEntry`

dehann, 2020-04-22:
Hi @Affie , cool thanks will do.  Next question, do you perhaps know the difference between FileDataStore and InMemoryDataStore (see links in first comment above)?

closed 2020-04-26 by dehann

dehann, 2020-05-02:
Again, I believe we decided against `BigData` as Noun and only using `Data` and `Smalldata` as Nouns.

PR #405 (2020-04-26): addDataEntry!,
fix #391
commit b8e256a4 (2020-04-26): addDataEntry!,

fix #391