JuliaRobotics/DistributedFactorGraphs.jl #400: [FEAT] listSolveKeys

dehann, 2020-04-24:
```julia
julia> listSolveKeys(fg)
:default
:parametric
:graphinit
:treeinit
:reference
```

Affie, 2020-04-25:
To fit with the rest of the API, I made a per variable list as: `listVariableSolverData(dfg::AbstractDFG, variablekey::Symbol)`
Dit you have a fg wide on in mind? Or should it maybe return a dictionary of solvekey=>label

dehann, 2020-04-26:
> `listVariableSolverData`

Oh cool thanks, did not know about that.

> Did you have a fg wide on in mind?

Yes, that was the idea -- would a "union" of all solveKeys available somewhere in the graph.  Havne't though on how to return that to user other just a vector.

> Or should it maybe return a dictionary of solvekey=>label

This could be one way to go yeah, where you get dict such that keys(dict) == listSolveKeys.  I think the idea with verb `list` is to always return a vector, right?   So we should use a different verb for returning dict of solveKeys and which variables contain them -- perhaps `buildSolveKeysDict`, or `findSolveKeysDict`?

Affie, 2020-04-26:
Remember to look here https://juliarobotics.org/DistributedFactorGraphs.jl/latest/ref_api/ for verbs implemented.

dehann, 2020-07-06:
I don't think we have resolved a union list return over the entire graph yet, so bumping to v0.9.0 -- if not we can close as part of v0.8.1.

Affie, 2020-07-07:
You can currenlty have a vector of vectors with broadcasting:
```juila
listVariableSolverData.(fg, ls(fg))
2-element Array{Array{Symbol,1},1}:
 [:default]
 [:default]
```
But no listSolveKeys yet.

dehann, 2020-07-19:
oh, thats right -- and easy:
```julia
listSolveKeys(dfg) = union(listVariableSolverData.(dfg, ls(dfg))...)
```

dehann, 2020-07-19:
and we already have the function, just tiding up in #538

closed 2020-07-19 by dehann