JuliaRobotics/DistributedFactorGraphs.jl #403: standardize API Entry or Element

dehann, 2020-04-25:
also see #401

Affie, 2020-04-25:
My intention with entry was to differentiate between just adding the ‘pointer’ to the data and the complete data record.

dehann, 2020-04-26:
HI @Affie , okay thanks -- could you perhaps update the Nouns in the wiki, not sure I follow 100% but we can use your lead as a starting point for the Nouns?

Maybe we can just add a clear example here of when to use Entry and when to use Element ...

# DF Guessing

Writing this without being sure:

## Example 1

I have a new piece of data I want to store, should I call `addDataEntry` or `addDataElement`?

## Example 2

I know there is a piece of data and trying to just get a copy and don't care exactly how it is retrieved -- `getDataEntry` or `getDataElement`

## Example 3

I'm trying to discover what data is available stores, should I be searching with the `listDataElements` or `listDataEntries`.

Affie, 2020-04-26:
Ok, will take a look. 
Why not just `addBigData!` for the wrapper version?

dehann, 2020-04-27:
Jip, that is what I did, see #405 -- and dont mind the error being fixed in #410 .

Although we still need to resolve `Entry` or `Element` for general use beyond the convenience wrapper.

dehann, 2020-05-02:
> Why not just addBigData! for the wrapper version?

Oh, should be more specific -- did we not decide against the wording "BigData" and opted for "Data" and and "Smalldata" Nouns instead?

Question here remains though, `Element` or `Entry` -- or at least what's the difference?

dehann, 2020-05-02:
So `Entry` is the "description" of a Data "Element".  The `Element` is kept in the `DataStore` while `Entry` description is kept in the DFG object?

dehann, 2020-05-02:
If so, then lines like these should be cleaned up:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/b8e256a4149794413c955a5c3a1fbe422dcd7d39/src/BigData/services/AbstractBigDataEntries.jl#L53

Affie, 2020-05-02:
It seems confusing and I don't really understand your question.

Why don't we name it `Key` and `Record`?
Or whatever the generic standard is for databases.

dehann, 2020-05-02:
> It seems confusing and I don't really understand your question.

I had hard time figuring out the difference between `Entry` and `Element` and seems the code is bit muddled.  My current understanding is that `Entry` is our own format to hold the DB key, description/label, mimeType etc:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/37bb5f481b7c44335770e7d13bff7176a90a5d51/src/BigData/entities/AbstractBigDataEntries.jl#L6-L12

<<<PS, `DataEntry` or `DataEntryGeneral` might be closer to the `verbNoun` naming convention -- we can sort that one out later.>>>

> Why don't we name it `Key` and `Record`?

I think `Key` does not envelope enough, since that would only be a unique identifier.  But these are related since we already have `entry.key` see above.

The question of `Element` vs "Record" is bit less clear cut.  Unfortunately "Record" and `Entry` sound very much the same too.  Currently it seems the code has `Element` playing role of a "Record".

My suggestion is we leave "Key" and "Record" as names used by the database technology used, in much the same way we have kept the name Vertex (perhaps Node) as name associated with the graphing technology underneath.

Lets hear if @GearsAD has an opinion on naming convention.

My suggestion for the short to medium term remains: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/403#issuecomment-622977797

dehann, 2020-05-02:
add added aliases for `addData!` and `fetchData` as was suggested above.

Affie, 2020-05-04:
Works for me, when I came up with the `Entry` thing I just named it the first word that came to mind, but I wanted a distinction from the then used `Element`. I don't really mind what its called and think Sam will know best as you suggest. As long as the functionality is there to add the "key" separately to the graph.

dehann, 2020-05-30:
Sounds good, think we are on the same page -- thanks for adding the `Entry` and `Element`, it will find its rhythm soon enough.  In an attempt to get right of confusion, how about one Noun from `Entry/Element/Record` (which seem quite similar to me personally),  and then pick either `Data/Blob` as a replacement for what is currently called `Element`?  I'm just thinking out loud -- this is likely a DFG v0.9.0 thing.

Affie, 2020-05-31:
From them I like `entry` and `blob`
So then
- a dataEntry refers to the `key+`
- a dataBlob revers to the `value` that will be at some `key+`

So something like `addDataBlob(dfg, label, entry=>blob)`

`getDataBlobs(dfg, label)`

GearsAD, 2020-06-08:
Yes, `entry` is what is saved in database (it's the details about the data) and `element` is the actual data itself (part that's saved on disk/etc.). I understand there's a bit of confusion about this because the terms are so similar. `entry` and `blob` are good.

dehann, 2020-06-17:
`entry=>blob` sounds great.  It's a 3-0 vote, thanks!

PR #519 (2020-07-06): BigData rename
- close #403 : rename Elements->Blob
- BigData -> Data
closed 2020-07-06 by Affie