JuliaRobotics/DistributedFactorGraphs.jl #404: updateBigDataEntry! should not be allowed?

dehann, 2020-04-26:
Did we decide that data entries are immutable and should not updatable?

This was big enough to be dubbed the "Falmouth Manifesto" -- is that

dehann, 2020-04-26:
perhaps a better verb here would be `replaceDataEntry!`?

Affie, 2020-07-09:
Perhaps DFG "core" should just remain vanilla with normal CRUD. The smarter functionality can then be added in the next layers such as CloudGraphsDFG for example.

dehann, 2020-08-12:
what is the different between `updateData!` and `updateDataEntry!` ?

Affie, 2020-08-18:
> what is the different between updateData! and updateDataEntry! ?

`updateData!` updates a data blob and entry
`updateDataEntry! ` updates a data entry

Affie, 2024-08-28:
Only blobs are currently immutable, BlobEntries are mutable, therefore update should be allowed. 
For example, one can update the metadata.

Affie, 2025-06-25:
Note: update will become merge