JuliaRobotics/DistributedFactorGraphs.jl #425: Abstract packing functions interface in DFG

Affie, 2020-05-01:
So far `decodePackedType` looks like the only one that has to be done.

GearsAD, 2020-05-01:
This is a good idea :+1:

Affie, 2020-05-03:
I missed `decodePackedTypeFunc` in the CGDFG struct, however it is not used. Abstract still seems like the way to go. I'll implement and then we can see how it works.

dehann, 2020-05-05:
HI @Affie , could you perhaps add some detail to the first comment about the issue here and #433 -- I'm not quite following?

Affie, 2020-05-05:
`decodePackedType` Is defined in IIF, but only used in one place in DFG. Its called in `Main` from DFG. 
I wanted to make a `decodePackedType` in DFG that IIF could add a method to, but it ended up not being possible with the current implementation. 
The only thing `decodePackedType` in DFG needs from IIF is the type to pack into, i.e. `CommonConvWrapper`. I didn’t want to get that from `Main` and it seemed reasonable to store it in the sover parameters.

dehann, 2020-05-05:
My understanding is `decodePackedType` has already been totally deprecated (but perhaps I'm mistaken)...  I actually did the work a while back to remove hard coded types/functions completely. 

> `decodePackedType` Is defined in IIF, but only used in one place in DFG

assume you mean this one:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/5d62dc7f7542ae63075f16ee4bf99a8d0fe413a1/src/services/Serialization.jl#L135

I thought the lifting was supposed to be done here?

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/69266b45e8cc1a7bd80ae3e42535684219cd485c/src/DispatchPackedConversions.jl#L41-L59

and here:

https://github.com/JuliaRobotics/IncrementalInference.jl/blob/69266b45e8cc1a7bd80ae3e42535684219cd485c/src/DispatchPackedConversions.jl#L68

Although now that I'm looking closer it seems this second link approach was lost in translation and we are back at the hard coded method I had previously removed ... :-P  I'm trying to check and make sure but seems like it at this point.

## How To Fix

Replace this:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/5d62dc7f7542ae63075f16ee4bf99a8d0fe413a1/src/services/Serialization.jl#L135

with this a simple convert that will use this:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/69266b45e8cc1a7bd80ae3e42535684219cd485c/src/DispatchPackedConversions.jl#L68

This approach is the precise mirror image of what is currently used in the packing path, here:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/5d62dc7f7542ae63075f16ee4bf99a8d0fe413a1/src/services/Serialization.jl#L104

and here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/69266b45e8cc1a7bd80ae3e42535684219cd485c/src/DispatchPackedConversions.jl#L65

## Must be Moved to DFG

These two functions should be moved to DFG:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/69266b45e8cc1a7bd80ae3e42535684219cd485c/src/DispatchPackedConversions.jl#L64-L69

Affie, 2020-05-05:
> how to fix

That is only part of it and done in DFG already if I'm following correctly.

Affie, 2020-05-06:
we can revisit but I think the rest are using convert

Affie, 2020-05-06:
forgot about `rebuildFactorMetadata!`

PR #449 (2020-05-07): Also dispach on solverData type for rebuildFactorMetadata!
Close #425
closed 2020-05-07 by Affie