JuliaRobotics/DistributedFactorGraphs.jl #432: tag v0.7.6

dehann, 2020-05-02:
@JuliaRegistrator register branch=release/0.7

JuliaRegistrator, 2020-05-02:
Error while trying to register: "Branch `release/0.7` not found"

dehann, 2020-05-02:
@JuliaRegistrator register branch=release/v0.7

JuliaRegistrator, 2020-05-02:
Registration pull request created: [JuliaRegistries/General/14059](https://github.com/JuliaRegistries/General/pull/14059)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.6 -m "<description of version>" 93c997329ed0ce4f1468c20741e9e7a06463e2e1
git push origin v0.7.6
```

closed 2020-05-02 by dehann