JuliaRobotics/DistributedFactorGraphs.jl #434: Possible compareAll bug

Affie, 2020-05-03:
```
Expression: compareFactorGraphs(fg, fg2, skipsamples = true, skipcompute = true, skip = [:initialized; :inferdim; :ppeDict; :solvedCount])
  StackOverflowError:
  Stacktrace:
   [1] eval at ./boot.jl:331 [inlined]
   [2] eval at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/DistributedFactorGraphs.jl:11 [inlined]
   [3] compareField(::Core.TypeName, ::Core.TypeName, ::Symbol) at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/services/CompareUtils.jl:40
   [4] compareFields(::Core.TypeName, ::Core.TypeName; show::Bool, skip::Array{Symbol,1}) at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/services/CompareUtils.jl:56
   [5] compareAll(::Core.TypeName, ::Core.TypeName; show::Bool, skip::Array{Symbol,1}) at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/services/CompareUtils.jl:137
   [6] compareAll(::Type{Any}, ::Type{Any}; show::Bool, skip::Array{Symbol,1}) at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/services/CompareUtils.jl:146 (repeats 6686 times)
   [7] compareFactorGraphs(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::LightDFG{SolverParams,DFGVariable,DFGFactor}; skipsamples::Bool, skipcompute::Bool, skip::Array{Symbol,1}, show::Bool) at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/src/services/CompareUtils.jl:388
   [8] top-level scope at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/test/iifCompareTests.jl:65
   [9] top-level scope at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.4/Test/src/Test.jl:1113
   [10] top-level scope at /home/travis/build/JuliaRobotics/DistributedFactorGraphs.jl/test/iifCompareTests.jl:9
  ```

dehann, 2020-05-03:
A stackoverflow like this might mean a dispatch is calling the same function endlessly.

dehann, 2020-07-06:
Stackoverflow has usually for me been a function recursively calling itself endlessly

Affie, 2020-07-07:
I'm going to leave this for now. I have some unfinished ideas on `compare` somewhere that will probably replace this.