JuliaRobotics/DistributedFactorGraphs.jl #441: REMOVE decodePackedType from solverParams

dehann, 2020-05-05:
See description of required work here: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/425#issuecomment-624225618

dehann, 2020-05-05:
packing and unpacking during serialization should not store types or functions for serialization!  Methods should rather be implemented as converters and operate in context Main.

Affie, 2020-05-05:
Just to be clear, `convert` is used, but I stored the wrapper type to convert to in solverParams. It made kind of sense since perhaps other algorithms would want to use different wrappers.

dehann, 2020-05-05:
> perhaps other algorithms would want to use different wrappers

Overloading the converters using multi-dispatch will achieve the same thing, no?

also:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/440/files#r420369361

Affie, 2020-05-05:
> Overloading the converters using multi-dispatch will achieve the same thing, no?

I wasn't able to find a way. but wanted to carry on so went with what I thought was best. Also I did't like `IncrementalInference.CommonConvWrapper` hardcoded in DFG.

dehann, 2020-05-06:
This is easily fixed like so:
- f62ef858f
- https://github.com/JuliaRobotics/IncrementalInference.jl/pull/695/commits/0a53db115e286b995ff20c7b16d7304043a3c04b

dehann, 2020-05-06:
> Just to be clear, convert is used, but I stored the wrapper type to convert to in solverParams

Hi @Affie , thanks -- after working through this and getting my head back in this space (took awhile) i see that is done, thanks for keeping that line going!

Turns out my hangup was misplaced in name but still valid in spirit -- but I feel a fix will quite easy to avoid storing the type (as with new fix linked above).

commit f62ef858 (2020-05-06): fix #441
PR #442 (2020-05-06): convenience buildSubgraph, fix #441
closes #441
commit 9adc57a8 (2020-05-06): Merge pull request #442 from JuliaRobotics/maint/2Q20/buildsfg

convenience buildSubgraph, fix #441
closed 2020-05-06 by dehann