JuliaRobotics/DistributedFactorGraphs.jl #453: DFGNodeParams vs, RefValue

Affie, 2020-05-10:
DFGNodeParams currently just has solvable in it.
```julia
mutable struct DFGNodeParams
    solvable::Int
end
```

If we are a planning more fields it might be worth keeping it, but will Variable and Factor share them or will they need VariableParams and FactorParams?

I don't like the complications for a maybe, therefore, for 1 or 2 fields I would prefer using a `RefValue` rather than `_dfgNodeParams` ie. `solvable::Base.RefValue{Int}`

dehann, 2020-05-13:
I am starting to use ::Ref{T} more myself (with usage e.g. `x[] = 1`) and like it.  I think the `DFGNodeParams` might have been to to get more structs as immutable -- remember taking to @GearsAD about it at some point.  If that was the case, I agree with Johan and lets drop this additional struct and move towards `solvable::Ref{Int}` for DFG v0.8.0.

Thanks for bringing this up and pinning to DFG v0.8.0!

dehann, 2020-05-13:
oh sorry, what is the difference between `RefValue` and `Ref` -- i misread that at first?

Affie, 2020-05-13:
`Ref` is an abstract container, so might have performance issues.

dehann, 2020-05-14:
Oh, thats significant -- thanks!

closed 2020-05-16 by Affie