JuliaRobotics/DistributedFactorGraphs.jl #462: do we need this warning?

dehann, 2020-05-14:
```julia
┌ Warning: Retrieving serialization module from AbstractDFG - override this in the 'LightDFG{SolverParams,DistributedFactorGraphs.DFGVariable,DistributedFactorGraphs.DFGFactor} structure! This is returning Main
└ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/LKg9S/src/services/Serialization.jl:201
```

dehann, 2020-05-14:
I dont even think the error message is valid?  Factor types are not available in LightDFG module, so why overwrite an unpacking function there?

dehann, 2020-05-14:
I'm getting thousands of printouts of this message when loading a FileDFG and not sure how to fix.

Affie, 2020-05-14:
I think it’s this one: #141

Affie, 2020-05-15:
Btw, it comes from this function: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/d7524b3e9d7b05dbb0392000e1a05820f68310a7/src/services/Serialization.jl#L200-L203

Affie, 2020-05-16:
Decision was to remove functions completely and use Main directly. So, no

commit 244d6b25 (2020-05-16): Remove get/setSerializationModule and always use Main
close #462
commit b639b3ce (2020-05-16): Remove get/setSerializationModule and always use Main
close #462
PR #467 (2020-05-16): Remove get/setSerializationModule and always use Main
close #462
commit 5efa24da (2020-05-16): Remove get/setSerializationModule and always use Main
close #462
closed 2020-05-16 by Affie