JuliaRobotics/DistributedFactorGraphs.jl #465: should lsfTypes just return a list of the types

Affie, 2020-05-15:
eg. [PriorPose2, Pose2Pose2, Point2Point2, ...]

It is currently a dictionary with type::Symbol=>Vector{type::String} whitch is a bit strange
```julia
Dict{Symbol, Array{String,1}} with 2 entries
:TestFunctorInferenceType1 => 
    Vector{String} with 1 element
        "TestFunctorInferenceType1"
:TestFunctorSingleton => 
    Vector{String} with 1 element
        "TestFunctorSingleton"
```

dehann, 2020-05-16:
HI, the idea a while back was it would return the types and which factor labels were of that type (hence the dictionary).  This was reasonable since the function lsType already had the labels available during the calculation, but no longer fits the verbNoun 100%.  So how about if we write the main function as `lsTypesDict` and then a simplified `lsTypes` more like a wrapper that only returns a list of types?

dehann, 2020-05-16:
should probably make ti:
```julia
Dict{Symbol,Vector{Symbol}}
```

and `lsTypes(args...; kwargs...) = keys(lsTypesDict(args...; kwargs...)) |> collect`

commit 88e743d4 (2020-05-18): implemet lsTypesDict, lsfTypesDict and test
close #465
closed 2020-05-18 by Affie