JuliaRobotics/DistributedFactorGraphs.jl #470: tag v0.7.7

Affie, 2020-05-17:
@JuliaRegistrator register branch=release/v0.7

JuliaRegistrator, 2020-05-17:
Error while trying to register: "Tag with name `v0.7.6` already exists and points to a different commit"

Affie, 2020-05-17:
@JuliaRegistrator register branch=release/v0.7

JuliaRegistrator, 2020-05-17:
Registration pull request created: [JuliaRegistries/General/14893](https://github.com/JuliaRegistries/General/pull/14893)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.7 -m "<description of version>" ae179ede7bc574279919e1da51a69de9fcb71041
git push origin v0.7.7
```

closed 2020-05-17 by Affie