JuliaRobotics/DistributedFactorGraphs.jl #474: add nanoseconds to DFGVariable and DFGFactor

dehann, 2020-05-19:
Julia `DateTime` dispatch only computes down to millisecond, but we frequently want micro or nanosecond.  To be fully compatible with ROS we should probably add a nanosecond offset memory location to.  I'd like to get this in for DFG v0.8.0.

dehann, 2020-05-19:
Suggested mechanism is to remove all subsecond from the `timestamp` field and store that separately at `nanosecond::Int64` scale (similar to ROS).  We'd need to do some logic to make it easy to use too.

Affie, 2020-05-19:
Is issue a duplicate of #367?
I was looking at it and best i could think of was to use `UTInstant{T}`.
I tend to agree with adding it in solver data as there is no specific standard time. 
The original idea behind variable timestamp was more for a standard time for things like database and searching and not as a variable for calculations.

GearsAD, 2020-05-22:
Current DB is ISO8601 with .fff, like timestamp:2020-05-22T08:38:49.797. As discussed we'll add an additional Int64 field with nanosecond resolution for the subsecond component, and additional functions to get/set/ the high-res timer.

PR #478 (2020-05-26): Feature add nanosecond time to DFGVariable and DFGFactor
close  #474

TODO
- [x] Serialize
closed 2020-05-27 by Affie