JuliaRobotics/DistributedFactorGraphs.jl #475: solverParams as keyword

Affie, 2020-05-21:
A mix of `solverParams` and  `params` was used as keywords.
Standardize to `solverParams`

dehann, 2020-05-24:
Hi @Affie, could we make this a keyword deprecation warning and not an error perhaps please -- as per semver?
```julia
julia> fg = generateCanonicalFG_Kaess()
ERROR: keyword `params` is deprecated, please use solverParams
Stacktrace:
 [1] error(::String) at ./error.jl:33
```

dehann, 2020-05-24:
thats also why this is failing:
https://travis-ci.org/github/JuliaRobotics/IncrementalInference.jl/jobs/690377230

Affie, 2020-05-24:
I’ll update to warning. Just note that cgdfg didn’t have params as keyword so I’m not doing anything there. 

> thats also why this is failing:
> https://travis-ci.org/github/JuliaRobotics/IncrementalInference.jl/jobs/690377230

Its testing again master and will only pass once merged to master. 
Note: I also update RoME accordingly.

dehann, 2020-05-24:
Got it thanks!

PR #479 (2020-05-23): standardise to solverParams
fix #475
commit b3e40835 (2020-05-23): standardise to solverParams
fix #475
closed 2020-05-24 by dehann