JuliaRobotics/DistributedFactorGraphs.jl #477: Validate Packing of `fnd.hypothesis` and `fnd.certainhypo`

dehann, 2020-05-22:
serializing factor nodes has newer fields relating to multihypothesis features and the serialization might be brittle or poorly organized in code.  Validate and rework these factor conversions in IIF.

Affie, 2020-05-22:
possibly related [IIF #699](https://github.com/JuliaRobotics/IncrementalInference.jl/issues/699)

dehann, 2020-05-24:
Yes I think so, shall we punt this one to DFG v0.8.1?

dehann, 2020-05-25:
On second thought, lets try fix for DFG v0.8.0...

Affie, 2020-06-01:
Just dumping related structures here to help:
```julia
mutable struct GenericFunctionNodeData{T<:Union{PackedInferenceType, FunctorInferenceType, FactorOperationalMemory}}
    eliminated::Bool
    potentialused::Bool
    edgeIDs::Vector{Int}
    fnc::T
    multihypo::Vector{Float64} # likely to moved when GenericWrapParam is refactored
    certainhypo::Vector{Int}
    solveInProgress::Int
end
```

```julia
struct DFGFactor{T, N} <: AbstractDFGFactor
    label::Symbol
    timestamp::DateTime
    nstime::Nanosecond
    tags::Set{Symbol}
    solverData::Base.RefValue{GenericFunctionNodeData{T}}
    solvable::Base.RefValue{Int}
    _variableOrderSymbols::NTuple{N,Symbol}
end
```

```julia
mutable struct CommonConvWrapper{T} <: FactorOperationalMemory where {T<:FunctorInferenceType}
  ### Values consistent across all threads during approx convolution
  usrfnc!::T # user factor / function
  # general setup
  xDim::Int
  zDim::Int
  # special case settings
  specialzDim::Bool # is there a special zDim requirement -- defined by user
  partial::Bool # is this a partial constraint -- defined by user
  # multi hypothesis settings
  hypotheses::Union{Nothing, Distributions.Categorical} # categorical to select which hypothesis is being considered during convolution operation
  certainhypo::Union{Nothing, Vector{Int}}
  # values specific to one complete convolution operation
  params::Vector{Array{Float64,2}} # parameters passed to each hypothesis evaluation event on user function
  varidx::Int # which index is being solved for in params?
  measurement::Tuple # user defined measurement values for each approxConv operation
  threadmodel::Union{Type{SingleThreaded}, Type{MultiThreaded}}
  ### particular convolution computation values per particle idx (varies by thread)
  cpt::Vector{ConvPerThread}
end
```

dehann, 2020-06-04:
> is hypothesis = multihypo
   
No they are different

> must nullhypo be added here to GFND?

Yes, should be added somewhere which will be persisted by serialization.

dehann, 2020-07-25:
FYI v0.9.0 adds `GFND.nullhypo::Float64`

Affie, 2020-08-27:
There are now 2 `nullhypo`s: In `CommonConvWrapper` and `GenericFunctionNodeData`
I just want to confirm they are 
- both needed 
- the same
I'm going ahead with the assumption they are the same.

Affie, 2020-08-27:
see https://github.com/JuliaRobotics/IncrementalInference.jl/pull/862

closed 2020-08-27 by Affie