JuliaRobotics/DistributedFactorGraphs.jl #482: oops, params keyword still an error

dehann, 2020-05-27:
got unsupported keyword argument "params"

```julia
MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  ...MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  ...MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  ...MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  ...MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  ...MethodError: no method matching LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,DFGVariable,DFGFactor}; params=SolverParams(0, nothing, nothing, false, 99999999999, false, false, true, true, true, false, false, 0.5, false, false, 500, 100, false, "/tmp/caesar/2020-05-27T13:14:38.022", true, false, [:default, :parametric], 3.0, Dict{Symbol,String}()))
Closest candidates are:
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::DistributedFactorGraphs.LightDFGs.FactorGraphs.FactorGraph{Int64,V,F}; description, userId, robotId, sessionId, userData, robotData, sessionData, solverParams) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
[... cut ...]
  LightDFG{SolverParams,DFGVariable,DFGFactor}(::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any, ::Any) where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:11 got unsupported keyword argument "params"
  LightDFG{SolverParams,DFGVariable,DFGFactor}() where {T<:AbstractParams, V<:AbstractDFGVariable, F<:AbstractDFGFactor} at /home/dehann/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:41 got unsupported keyword argument "params"
```

Affie, 2020-05-27:
I can't reproduce.

dehann, 2020-05-27:
Oh, I'm one commit behind master on a feature branch.  Will need to do some creative fast-forwarding, but should be fine.

closed 2020-05-27 by dehann