JuliaRobotics/DistributedFactorGraphs.jl #494: Possible loadDFG issue

dehann, 2020-06-06:
ah, i just came across a weird one on all masters:
```julia
fg = initfg()
laodDFG(joinpath(...), Main, fg)

┌ Warning: `loadDFG(source::String, iifModule::Module, dest::AbstractDFG)` is deprecated, use `loadDFG!(dest, source)` instead.
│   caller = top-level scope at DebuggingWineGlass.jl:7
└ @ Core ~/work/mit/papers/iros2020_affordances/code/DebuggingWineGlass.jl:7
```

but when i try:
```julia
fg = initfg()
loadDFG(fg, joinpath(...))

MethodError: no method matching loadDFG(::LightDFG{SolverParams,DistributedFactorGraphs.DFGVariable,DistributedFactorGraphs.DFGFactor}, ::String)
top-level scope at DebuggingWineGlass.jl:9```
<sub>[View in Slack](https://caesarjl.slack.com/archives/CKGRFPZT8/p1591456909000900)</sub>
```

Affie, 2020-06-06:
it mutated fg so i added the `!` `loadDFG!`

dehann, 2020-06-06:
Ah thanks, I'll perhaps add another deprecate to help the transition.

closed 2020-06-08 by dehann

dehann, 2020-06-09:
right, so part of the issue was that when using in RoME and Caesar, IIF would be reexported by them but not DFG.  IIF then exports the loadDFG on DFG's behalf.  Problem was that loadDFG! was not defined in IIF...

PR #495 (2020-06-08): help with #494 cases
close #494
commit e7fcedba (2020-06-08): help with #494 cases

close #494