JuliaRobotics/DistributedFactorGraphs.jl #507: CGDFG -> local fails with cryptic error

GearsAD, 2020-06-11:
If I have CGDFG object with `NoSolverParams` and I create a local LightFG with `SolverParams`, when I try use `copyGraph!` it fails with "Please extend your workspace with function getFactorOperationalMemoryType(<:AbstractParams) for your usecase, ..."

Is that expected? I would think if there is an error there it should be that your SolverParams don't match, although I don't think that should stop copying the graph.

Affie, 2020-06-11:
I tried it, but I cannot reproduce. However, I think I know what scenario might cause the error. 
`getFactorOperationalMemoryType` has to be defined on subtype `NoSolverParams` to create a cloudgraph, this is only done in the tests as it didn't make much sense beyond testing to work with `NoSolverParams`. If you remember previously CloudGraphs only worked with IIF. 
Can you try to define `getFactorOperationalMemoryType(p::NoSolverParams)` and the functions that go with it for your use case and see if it works?

Affie, 2020-06-11:
Also, note that it will not work in the solver to copy from one NoSolverParams to SolverParams with the test definitions. 
My suggestion would be to move the definition of `NoSolverParams` to be included only in testing and then to look at copying from one type of solverParams to another when a use case for it arises.

GearsAD, 2020-06-12:
:+1: will try over the weekend

dehann, 2020-06-14:
> object with NoSolverParams and I create a local LightFG with SolverParams, when I try use copyGraph! it fails

Touches on #264 .  I think the desired behavior should make it possible to copy from SolverParams to a NoSolverParams object (better cross-feeding).

> it will not work in the solver to copy from one NoSolverParams to SolverParams

So that could a problem, because that is exactly the use case we need for solving database versions of the graph -- the original reason for building DFG.  Sounds like some kind solution will be needed here for sure.

> copying from one type of solverParams to another

Yes exactly, directly relates to #264.

> move the definition of NoSolverParams to be included only in testing

That might not be possible since DFG should be able to stand on its own?  SolverParams is defined in IIF so some stopgap is needed in DFG.  I have been having trouble with NoSolverParams thing too, and been getting around it with lazy constructor `initfg()` and always just using that, however, now is best time to either change or augment the DFG design if necessary.

Affie, 2020-06-14:
Hi @dehann 
It's not a bug. If you want copy and seriailizaiton functions to work for `NoSolverParams` it has to be implemented. So rather make a feature request and describe what you need out of it. It should be possible within the existing framework.
We can create a 'TestSolverParams' type for tests and then add serialization for `NoSolverParams` in DFG that is compatible with IIF, but why not use IIF directly? 

> been getting around by always using lazy in-memory constructor initfg()

I think that's the way to go for most users. I've been using the constructors directly (without issues), but that's because I use experimental features. 
Also, DFG docs recommend using the IIF versions of addVariable! and addFactor so it is solvable for the same reasons.

dehann, 2020-06-15:
> DFG that is compatible with IIF, but why not use IIF directly?

Agree, that is the lesser of the issues so go that way if necessary.  It would be nice if DFG supports #264 on its own, if possible with time we have.

> make a feature request and describe what you need out of it

"""
If I have CGDFG object with NoSolverParams and I create a local LightFG with SolverParams, when I try use copyGraph! it fails
"""

I would say that is a feature request that should work?  I agree with Sam: "Is that expected? I would think if there is an error there it should be that your SolverParams don't match, although I don't think that should stop copying the graph."

So to achieve #264 with specific regard to varied <:AbstractSolverParam in each of the different dfg objects, we would need to implement stubs for these funcitons? 

> Can you try to define getFactorOperationalMemoryType(p::NoSolverParams) and the functions that go with it

closed 2023-04-19 by Affie