JuliaRobotics/DistributedFactorGraphs.jl #509: Smalldata, how about fake `Entry` as wrapper?

dehann, 2020-06-17:
Quirky decision, how about dropping small data dict framework, and only maintaining the `Data` plumbing but allow entries that do not have an associated `Blob`?  We can flag the `Entry` object as having a `Blob` or not as immutable field?

cc @GearsAD @Affie

GearsAD, 2020-06-17:
Not a bad idea, I'm about to separate out the big data as separate nodes though, so that may make this difficult.

GearsAD, 2020-06-17:
I guess we could still do it, we would just have to collect the entries around the node, but it would be a bit more difficult than a JSON blob inside the actual node.

Affie, 2020-06-17:
It fits in with why I originally created the `AbstractBigDataEntry`  (think to be renamed `AbstractDataEntry`) in our application.
I just stored the data in a custom structure directly. For example, tags were something like:
```julia
struct TagData <: AbstractDataEntry
    key::Symbol
    family::String
    tagids::Vector{Int}
    bTt
end
```
The problem was a concern that the payload could get too big. That's why the in-memory data store was created. I don't like and never used `smalldata` because of the `String`, but it can perhaps be consolidated as:
```julia
struct SmallData <: AbstractDataEntry
    key::Symbol
    data::String
   #OR 
    smallDataDict::Dict{String, String}
end
```

dehann, 2020-07-28:
Affects JuliaRobotics/Caesar.jl#514

dehann, 2023-02-28:
we could if we really wanted to store something like this using a combo of the `BlobEntry.description`, `BlobEntry.metadata`, and `BlobEntry.mimeType` fields.

closed 2023-02-28 by dehann

dehann, 2023-03-01:
Also,
```julia
julia> UUID("00000000-0000-0000-0000-000000000000")
UUID("00000000-0000-0000-0000-000000000000")
```