JuliaRobotics/DistributedFactorGraphs.jl #520: DataEntry should be refactored

GearsAD, 2020-07-03:
For 0.9 we should refactor the data entries to include more information.

The proposed data entry structure that every data source (e.g. MongoDB) should implement is:
```julia
struct AbstractDataEntry
  id::UUID
  label::Symbol # Formerly key
  source::String # E.g. user|robot|session|label
  description::String
  mimeType::String
  hash::String # Probably https://docs.julialang.org/en/v1/stdlib/SHA
end  
```

We can't actually implement this in `AbstractDataEntry`, but we should enforce that these fields are populated.

dehann, 2020-07-03:
i would suggest we don’t call this AbstractDataEntry, since Abstract has specific meaning in our general dispatch use.  Why not just DataEntry and it can derive from AbstractDataEntry if that abstraction is required?

dehann, 2020-07-03:
rest looks good to me!

GearsAD, 2020-07-03:
Agreed, that was more of a conceptual design.

In practice these fields would have to be populated in `GeneralBigDataEntry`, `MongodbBigDataEntry`, and `FileBigDataEntry`.

dehann, 2020-07-05:
guessing you mean “Data” :-)

Affie, 2020-07-07:
xref https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/97#issuecomment-537984850

Affie, 2020-07-09:
What hashing function do we want to use and will it be fixed to only one?
Julia is currently faster on SHA2

Affie, 2020-07-09:
Do we want a timestamp?
The variable has a timestamp that should technically be the same (or close to) the data timestamp.
Perhaps it can help with data integrity.

GearsAD, 2020-07-13:
SHA2 is good I think?
The timestamp is more of a database-level entry, just to keep track of when the data was actually created, the timestamp in the variable is still the authority on the variable.

GearsAD, 2020-07-13:
Notes from Slack:
- Hash is base64 encoded string
- We'll (optionally) use datasource ID to identify the different stores, which we'll save into the entries
- Users are responsible for managing the different stores
- Data is immutable, so we don't have to worry about updateTimestamp or updateData()
- Line 115 with addData!(dfg::AbstractDFG, dataStore::AbstractDataStore, variableLabel::Symbol, dataLabel::Symbol, blob::Vector{UInt8}, [default parameters like description etc.]) will return the entry when you create the data (really like that way of working with the API as you have it there)
- SmallData is a special data entry as you've got there, of type Dict{Symbol, Union{Int, Float64, AbstractString, Vector{Int}, Vector{Float64}, Vector{AbstractString}} or something similar
- SmallData is a special DataEntry, but it's only really used in CGDFG for saving/loading. A user won't see it, the DFGVariable will contain the Dict like above

dehann, 2020-07-13:
I have a suggestion on this:

> We'll (optionally) use datasource ID to identify the different stores, which we'll save into the entries

The idea behind immutable id => blob pairs is that the location of the blob does not matter.  in fact its highly likely that there will be multiple copies of the blob flying all over the place (hence the immutable requirement), so it seems little awkward to store a single blob store location into the Entry?

dehann, 2020-07-19:
targeting v0.9 or v0.10 now that we pushing for nullhypo stuff in v0.9?

closed 2020-07-28 by Affie