JuliaRobotics/DistributedFactorGraphs.jl #523: Use Time Zone

Affie, 2020-07-04:
We can either use 
- `ZonedDateTime` or
- Add a `TimeZone` field

DateTime is used extensively in the system, we can change to a time zone by:
- assume `UTC` for time zone
- assume `Local` for time zone

`now()` returns local time, so we could probably safely assume Local time.

My vote would be for:
- `ZonedDateTime` 
- assume `Local` for time zone

dehann, 2020-07-05:
My suggestion is a deprecation cycle and end with whatever is the better design.  

I also want to suggest keeping the option for accurate nanosecond info available in the timestamp — that can be sorted out in future version, but please keep that door open at this stage.  ROS uses timestamp and a separate nanosecond offset containing the subsecond information, ie (timestamp mod second) + nanosecond/1e9

Affie, 2020-07-19:
So the idea is that set methods will remain flexible with the use of DataTime, but assuming local time zone as in:
```julia
setTimestamp(v::DFGVariable, ts::DateTime, timezone=localzone(); verbose::Bool=true) = setTimestamp(v, ZonedDateTime(ts,  timezone); verbose=verbose)
```

Affie, 2020-07-19:
With the current serialization, the time zone name (eg. Africa/Johannesburg) is not maintained only the +2.00.

dehann, 2020-07-20:
sounds good to me!

Affie, 2020-07-20:
> With the current serialization, the time zone name (eg. Africa/Johannesburg) is not maintained only the +2.00.

Also just note the loss of daylight saving time.

PR #543 (2020-07-20): Change DateTime to ZonedDateTime
close  #523
closed 2020-07-20 by Affie