JuliaRobotics/DistributedFactorGraphs.jl #527: Remove all return types?

dehann, 2020-07-06:
As before, we are removing return types, correct?

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/master/src/services/AbstractDFG.jl

dehann, 2020-07-06:
From Slack:

dehann
Johan and i voted, 2-0 for no return types :-). the reason is that return types in Julia work different from C.  The most troubling is when you convert a healthy concrete type up to an abstract type, which is happening a lot in the code at the moment, then you loose all the static operation and replace good code with dynamic dispatch around the forced abstract types.  Furthermore, forcing a return type inserts a convert function in all cases which ends up producing really difficult error messages on a failure.  So using return types doesn’t buy you anything in terms of type stability, it actually takes away.  The way to identify type issues is with @warn_type

@Affie 
I’ve been taking out the abstract return types, but not the rest. Are we removing all of them? Ideally the convert should be optimized out and asset can be useful, but it does lead to difficult error messages. The c programmer in me still likes it and there are places that it helps. I agree it helps with understanding the code. So maybe we should move it to the doc strings where it’s not needed.
I don’t think @warn_type will do anything. Calling assert on the return type is a bit closer.
We have also been changing arrays to abstract arrays and floats64 to real in a lot of places to allow more flexibility. It was needed for automatic differentiation in my case.

dehann, 2020-07-06:
> So maybe we should move it to the doc strings where it’s not needed.

Agree, at minimum lets do docstrings.

> Calling assert on the return type is a bit closer.

That would be for the developer only, and likely reduce performance a bit, so maybe a comment line with the assert in (at the end of a function) so when developing it can be uncommented if needed?

> It was needed for automatic differentiation in my case.

oh thats news to me, by "it" do you mean the hard return type was needed?

For Julia in general, a function should arrive at a type stable solution (which is where `@warn_type` helps).  Forcing a return type does not help the type inference as far I'm aware.

Affie, 2020-07-06:
> oh thats news to me, by "it" do you mean the hard return type was needed?

no, the opposite. It had to be abstract input types to work.

> That would be for the developer only, and likely reduce performance a bit, so maybe a comment line with the assert in (at the end of a function) so when developing it can be uncommented if needed?

@assert is a bit more complicated, from the docs:
> An assert might be disabled at various optimization levels. Assert should therefore only be used as a debugging tool and not used for authentication verification (e.g., verifying passwords), nor should side effects needed for the function to work correctly be used inside of asserts.

Affie, 2020-07-07:
BTW, annotating return types calls:
```julia
typeassert(x, type)
Throw a TypeError unless x isa type. The syntax x::type calls this function.
```

dehann, 2020-07-07:
Hi Johan, thanks that is good info.  Sounds like you are still opposed to using return types (even though your C sensibilities guide you otherwise)?

sam-globus, 2020-07-07:
If the vote is 2 for, 1 against, I'll clean up my return types, but *spooky music* I foresee danger with this change, no good things have come from type ambiguity.

Affie, 2020-07-07:
I'm a bit on the fence, I don't want the `convert` part, but like the `typeassert` part for development.

If you do this:
```julia
foo(x::Real)::Union{Nothing, Real} = x < 0 ? nothing : x
```
```julia
convert(Union{Nothing, Real}, nothing)
```` 
is called.

That just weird.

Affie, 2020-07-07:
the call to convert can also hide problems 

```julia
julia> function foo1(x::T) where T<:Real
           y = x * 10.0
           return y::T
       end
foo1 (generic function with 1 method)

julia> function foo2(x::T)::T where T<:Real
           y = x * 10.0
           return y
       end
foo2 (generic function with 1 method)

julia> foo1(10)
ERROR: TypeError: in typeassert, expected Int64, got Float64
Stacktrace:
 [1] foo1(::Int64) at ./none:3
 [2] top-level scope at none:0

julia> foo2(10)
100
```

Affie, 2020-07-07:
Something like this will be the bad case Dehann mentioned.
```julia
function foo(x::Real)::Vector{Real}
    return [x, x, x]
end
```

Affie, 2020-07-07:
xref #374

GearsAD, 2020-07-07:
Yep, makes sense that the implicit convert is the issue here, so it's not really type enforcing. Maak geluide soos n plan :+1:

closed 2020-07-19 by dehann