JuliaRobotics/DistributedFactorGraphs.jl #533: change Singleton / Pairwise to AbsoluteData / RelativeData

dehann, 2020-07-17:
Or something, we can discuss here

dehann, 2020-07-17:
https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-660138828

Affie, 2020-07-17:
The applicable code:
```julia
##==============================================================================
## Abstract Types
##==============================================================================

abstract type InferenceType end
abstract type PackedInferenceType end
abstract type FunctorInferenceType <: Function end
abstract type FactorOperationalMemory <: Function end
abstract type FunctorSingleton <: FunctorInferenceType end
abstract type FunctorPairwise <: FunctorInferenceType end
abstract type FunctorPairwiseMinimize <: FunctorInferenceType end
```
If a `FunctorSingleton` is always a prior, I would suggest 'Prior' then. It might be less of a learning curve to stick with familiar names? It took me a while to figure out what it was.

Also is the naming of `Functor` needed in  `FunctorInferenceType` and `FunctorSingleton`
maybe it can be reduced to 'AbstractPriorFactor` or even `AbstractPrior` and `AbstractRelativeFactor` ?

dehann, 2020-07-17:
Yes, these two sound best to me -- let's do that:

> `AbstractPrior` and `AbstractRelativeFactor`

GearsAD, 2020-07-19:
Personally I like simpler/shorter, `AbstractPrior` and `AbstractRelativeFactor` sound like a plan

dehann, 2020-07-19:
Lets do a deprecation cycle on the names because RoME and Caesar are both using Singleton and Pairwise abstracts at the moment.  So basically just create the new ones next to it and place those ones in the Deprecated.jl file.

PR #537 (2020-07-19): Update as per #533
Closes #533
closed 2020-07-19 by dehann