JuliaRobotics/DistributedFactorGraphs.jl #542: Save DFG version into serialized data

GearsAD, 2020-07-19:
This is to support the detection/migration of data from older versions

dehann, 2020-07-20:
This would be super!  Would there be like a manifest file or something?  

This might be related to the issue or separate user config information.  Just linking for future reference, don't want to hijack this thread beyond versioning.
- JuliaRobotics/IncrementalInference.jl#133
- JuliaRobotics/IncrementalInference.jl#143

GearsAD, 2020-08-15:
At the moment we save the version into the principal objects, and then we can look at supporting automatic migrations - "Oh this is v0.10 but we're at v0.11 so automatically call `_migrateDataObject_v010_v011_DFGFactor(...)`.

Good idea for the complete tar, will have to include in next release - #622

closed 2020-08-16 by GearsAD