JuliaRobotics/DistributedFactorGraphs.jl #554: Evaluation into the closed module `DistributedFactorGraphs` breaks (don't do this)

dehann, 2020-07-26:
```julia
[ Info: Including CloudGraphsDFG

ERROR: LoadError: InitError: Evaluation into the closed module `DistributedFactorGraphs` breaks incremental compilation because the side effects will not be permanent. This is likely due to some other module mutating `DistributedFactorGraphs` with `include` during precompilation - don't do this.
Stacktrace:
 [1] include(::Function, ::Module, ::String) at ./Base.jl:380
 [2] include at ./Base.jl:368 [inlined]
 [3] __init__() at /home/travis/.julia/packages/DistributedFactorGraphs/r6Be7/src/DistributedFactorGraphs.jl:295
 [4] _include_from_serialized(::String, ::Array{Any,1}) at ./loading.jl:697
 [5] _require_from_serialized(::String) at ./loading.jl:749
 [6] _require(::Base.PkgId) at ./loading.jl:1040
 [7] require(::Base.PkgId) at ./loading.jl:928
 [8] require(::Module, ::Symbol) at ./loading.jl:923
 [9] include(::Function, ::Module, ::String) at ./Base.jl:380
 [10] include(::Module, ::String) at ./Base.jl:368
 [11] top-level scope at none:2
 [12] eval at ./boot.jl:331 [inlined]
 [13] eval(::Expr) at ./client.jl:467
 [14] top-level scope at ./none:3
during initialization of module DistributedFactorGraphs
in expression starting at /home/travis/build/JuliaRobotics/IncrementalInference.jl/src/IncrementalInference.jl:15
ERROR: LoadError: LoadError: Failed to precompile IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480] to /home/travis/.julia/compiled/v1.5/IncrementalInference/NrVw2_kKNtl.ji.
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] compilecache(::Base.PkgId, ::String) at ./loading.jl:1290
 [3] _require(::Base.PkgId) at ./loading.jl:1030
 [4] require(::Base.PkgId) at ./loading.jl:928
 [5] require(::Module, ::Symbol) at ./loading.jl:923
 [6] include(::String) at ./client.jl:457
 [7] top-level scope at /home/travis/build/JuliaRobotics/IncrementalInference.jl/test/runtests.jl:6
 [8] include(::String) at ./client.jl:457
 [9] top-level scope at none:6
in expression starting at /home/travis/build/JuliaRobotics/IncrementalInference.jl/test/TestModuleFunctions.jl:7
in expression starting at /home/travis/build/JuliaRobotics/IncrementalInference.jl/test/runtests.jl:6
```

cc @GearsAD

dehann, 2020-07-26:
not sure if the is related, but remember if you want things to run in global scope you can use the `@eval include("....")` macro.

dehann, 2020-07-26:
It looks like this error only shows up if you do:
```julia
using IncrementalInference
```

but not on 
```julia
using DistributedFactorGraphs
```

My guess is something might have gone awry on #517  but have not been able to trace anything specific yet.  I might delay looking at this to focus on other stuff for the time being.

dehann, 2020-07-26:
Here is what was suggested by Tim Holy for a similar error:
- https://github.com/JuliaGraphics/ColorTypes.jl/pull/178#issuecomment-614902798
- JuliaLang/julia#35501

dehann, 2020-07-26:
Asking @Affie , if you perhaps have some bandwidth to look at this at some point perhaps please ( part of request in JuliaRobotics/IncrementalInference.jl#811 too)

GearsAD, 2020-07-27:
Thanks, yes this has been an issue for a while now, appreciate bringing it up. @jim-hill-r you mentioned this during one of the discussions as well, I think?

Affie, 2020-07-27:
I may have lost an eval in a merge at some stage, but it doesn't seem to matter: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/cbd5dc732966d1e4888d3742be4d9b2561fdc54e#diff-f0a6886e83d346a4ec106f2dd824a47c
I can't reproduce the error after #556.

Affie, 2020-07-27:
Ok, I got a similar error. There was a choice between 2 ways to include CGDFG. The other one was to include it with precompiling perhaps that's the better way then.

dehann, 2020-07-27:
Hi Johan,  just making sure you did see Tim Holy's suggested solution using `__init__` approach.  Thanks for taking a look!

Affie, 2020-07-28:
> Hi Johan, just making sure you did see Tim Holy's suggested solution using __init__ approach. Thanks for taking a look!
I have. It doesn't make a difference. 

The easiest way to fix it is to use some of the Requires.jl functions.
If we want precompilation then we can have the following outside the __init()__:
```julia
if get(ENV, "DFG_USE_CGDFG", "") == "true"
    @info "Detected ENV[\"DFG_USE_CGDFG\"]: Including optional CloudGraphsDFG (LGPL) Driver"
    include("CloudGraphsDFG/CloudGraphsDFG.jl")
    @reexport using .CloudGraphsDFGs
end
```
The caveat will then be that DFG has to be precompiled with DFG_USE_CGDFG=true and it stays in the cache. 
I haven't found a way to make this easy, that's why I added it to the __init()__ function.

dehann, 2020-07-28:
Okay, cool -- lets go with what @Affie suggests -- I just wanted to make sure you'd seen since Tim is a real expert on Julia.  Thanks, all good!

closed 2020-07-31 by Affie

dehann, 2020-08-08:
Also, breadcrum: GiovineItalia/Compose.jl#139