JuliaRobotics/DistributedFactorGraphs.jl #559: idea: addData!(...., datastore=SomethingNonDefault)

dehann, 2020-07-27:
Following #550 , I was wondering if it would be worth having an interface in the future that allows multiple datastore (with a single user default already in the `fg` object:
```julia
addData!(fg, :x1, data..)

# vs. also being able to immediately also push to an alternate datastore

addData!(fg, :x1, data.., datastore=myAltDatastore)
```

Affie, 2020-07-31:
is it not rather `copyDataBlob`? I also do not think an entry can be associated with more than one blob a time?

dehann, 2020-08-01:
> also do not think an entry can be associated with more than one blob a time

Ah, right wow -- that will take some thinking.  Lets work with single `Entry => Blob` assumption for the time being.  `copyDataBlob` functionality seems like a good workaround in the meantime.

Thoughts like caching blobs from a remote store in a local store also pop up.  Lets move this to a longer term discussion.  Thanks!