JuliaRobotics/DistributedFactorGraphs.jl #563: Docs: Improve Data Blobs and Entry docs

Affie, 2020-07-28:


dehann, 2020-07-28:
from #550

dehann, 2020-07-28:
For persistence: the FolderStore, for quick and easy (not persisted or copied to the cloud) I like the in-memory data entry.

We currently have:

dataset1 = rand(UInt8, 1000)

ade,adb = addData!(InMemoryDataEntry, dfg, :x1, :random, dataset1)
#or
ade,adb = addData!(FileDataEntry, dfg, :x1, :random, "/tmp/dfgFilestore", dataset1)
#or
ds = FolderStore{Vector{UInt8}}(:filestore, "/tmp/dfgFilestore")
addBlobStore!(dfg, ds)
ade,adb = addData!(dfg, :filestore, :x1, :random, dataset1)

The docs are a bit behind and we need good examples. The main purpose of PR was to have a clean API.

Affie, 2020-08-19:
Documented in https://juliarobotics.org/Caesar.jl/latest/concepts/interacting_fgs/#Adding-A-FolderStore

closed 2020-08-19 by Affie