JuliaRobotics/DistributedFactorGraphs.jl #565: MethodError:  updateVariableSolverData!

Affie, 2020-07-29:
MethodError: no method matching updateVariableSolverData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::Bool, ::Array{Symbol,1}, ::Bool)

from changes here as far as I can tel

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/15e57d5d9000194c7219523631a0d2e91a10c538/src/services/DFGVariable.jl#L434-L446

Affie, 2020-07-29:
Add tests for this in DFG

Affie, 2020-08-12:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/21655bf6bcda16fe1c9b1f099f5c653a5b827ac5/src/services/DFGVariable.jl#L524

We should decide how we want to use this.

Affie, 2020-08-12:
originally posted in a comment: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/570#discussion_r464301676

Maybe we shouldn't overwrite expected behaviour from base copy with deepcopy?
I expect `copy` to return a shallow copy and deepcopy for this.

Personally I don't think we should provide too many functions. Every function has to be documented and maintained.
The user can rather just do 
```juila
dcvnd = deepcopy(vnd)
dcvnd.selverKey = newKey
```

Is the function we are trying to replace `updateSolverData!(dfg, label, vnd, newsolvekey)`?
Where `vnd.solvekey != newsolvekey`.
Perhaps something like (although IIF needs to be updated slightly for it):
```julia
function deepcopySolverData!(dfg::AbstractDFG, label::Symbol, srcKey::Sybol, dstKey::Symbol)
    vnd = deepcopy(getSolverData!(dfg, label, srcKey))
    vnd.solverKey = dstKey
    updateSolverData!(dfg, vnd)
end  
```
Now that I'm thinking it through. `update` is maybe not the correct function to use. Its adding new solverdata so maybe:
```julia
function addSolverData!(dfg::AbstractDFG, label::Symbol, vnd, solverKey::Symbol)
    vnd.solverKey = dstKey
    addSolverData!(dfg, vnd)
end  
```

dehann, 2020-08-12:
> Maybe we shouldn't overwrite expected behaviour from base copy with deepcopy?

definitely not change the behaviour of a method from Base.

dehann, 2020-08-12:
> Is the function we are trying to replace updateSolverData!(dfg, label, vnd, newsolvekey)?

No, the idea is just to have the solveKey available inside the structure as a duplicate from the Dict container in which the VND is stored.  Nothing crazy, just duplication.

> Perhaps something like (although IIF needs to be updated slightly for it):

I disagree with the user having to do band-aids for DFG internal duplication decision.

dehann, 2020-08-12:
> update is maybe not the correct function to use. Its adding new solverdata so maybe:

`update` verb will need to update both registers of the duplication.

Affie, 2020-08-12:
> No, the idea is just to have the solveKey available inside the structure as a duplicate from the Dict container in which the VND is stored. Nothing crazy, just duplication.

Dehann, I think you are missing the point

dehann, 2020-08-12:
maybe, how I understand it is that Sam asked for a duplicate of the solveKey inside VND.  So nothing changes, except that there is a forced duplication of on ::Symbol register.

Affie, 2020-08-12:
No that is not the issue. The issue is it was used in IIF to copy from one solver data to another. and the key was removed from the parameters completely. I thought to add copy functionality.

dehann, 2020-08-12:
> I thought to add copy functionality.

not following

> The issue is it was used in IIF to copy from one solver data to another.

Do you perhaps have the piece of code where that is happening?

Affie, 2020-08-12:
I'll try and find it. it is where the init solve key is made

dehann, 2020-08-12:
Okay, so that line in IIF is using the `update` verb to insert a new copy.  So in that specific use case in IIF `update` is fine, and IIF is responsible for making a `deepcopy` of the `VND` data.

I would add---to address the duplication of solveKey in VND---that the parameter argument passed to the `update` function takes precedence.  So when a user updates a VND, then DFG should check that the duplication of registers matches, and discrepencies are not introduced:
```
vnd = getVND(..., :default)
dcvnd = deepcopy(vnd) # solveKey=:default
updateVariableSolverData!(... dcvnd, , :graphinit, ...)
```
In this case, DFG should update `dcvnd.solveKey = :graphinit`.

PS, I think `solverKey` is a spelling mistake too.

dehann, 2020-08-12:
remember the user of VND is unaware that there is duplication of `solveKey`.

Affie, 2020-08-12:
My and Sam's vote is for 

```julia
vnd = getVND(..., :default)
dcvnd = deepcopy(vnd) # solveKey=:default
vnd.solverKey = :graphinit
addVariableSolverData!(..., dcvnd)
```
Just because ist a little strange to have `updateVariableSolverData!` sometimes modifying your vnd.

Affie, 2020-08-12:
> PS, I think solverKey is a spelling mistake too.

I saw there was a bunch of solvekey and a few solverkey mixed. I think Sam went with solverkey

dehann, 2020-08-12:
i disagree, would rather have in weird cases:
```julia
addVariableSolverData!(..., dcvnd, checkSolvekey=false)
```

Because everything else about the entire world of coding is to avoid duplication.  There is a special interest inside the CGDFG driver to have this quirky duplication which should not be forced down on users.  If you don't know about the `VND.solveKey` duplication then you are bound to make mistakes.  If you know about the duplication and want to force a discrepancy, then use a special keyword to suppress the "automation".

dehann, 2020-08-12:
> I think Sam went with solverkey

Yeah, that needs to be fixed.

dehann, 2020-08-12:
okay, so i thought some more and came up with a compromise (especially since i’m our vote 2-1).  Just do both.  Do as Johan and Sam suggest by extracting VND.solveKey on update!.  Also write a second update! wrapper function with different parameter arguments which includes the ::Symbol which first updates the VND.solveKey assuming checkSolvekey and then calls the first function.

dehann, 2020-08-13:
FYI:

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/182a5b6246e33b0dd0c9388130fbe0b4b3b7a835/src/services/DFGVariable.jl#L552-L569

Affie, 2020-08-13:
`deepcopySolvekeys!` I didn't know about that one. (or forgot)  it similar to my suggestion of `deepcopySolverData`. I would say either use that or https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/565#issuecomment-673038085
No need to make another similar function.

Affie, 2024-08-28:
This is somewhat related to #1084, maybe both will use the same new verb.

Affie, 2025-06-26:
BTW, copying the whole VariableState as in copyto is faster than updating fields (used prevoiusly):
- `updateVariableSolverData!` 0.000626 seconds (1.11 k allocations: 114.289 KiB)
- `copytoVariableState!` 0.000099 seconds (315 allocations: 27.758 KiB)

dehann, 2025-07-01:
I think it's fine to use copyto.  Let's get real feedback and see if merge vs copyto comes up again for DFG v2.  Also the fallback of just aliasing if there is duplicity later.

closed 2025-07-01 by Affie