JuliaRobotics/DistributedFactorGraphs.jl #568: Should we revert softtype changes to deprecate cleanly?

Affie, 2020-07-30:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/63ec2eac7cfb390199cda96e761fc3e47cacef82/src/services/DFGVariable.jl#L41

We can take a smaller step as proposed in https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-660308134 (As said there, not with the Unions. Traits might be good.)

It's a big change that looks impossible to deprecate without breaking a lot of downstream code. 

I would propose that we first move all the state information out of the softtypes downstream and then make the change.

dehann, 2020-07-31:
> I would propose that we first move all the state information out of the softtypes downstream and then make the change.

Not sure I follow the move downstream part, but I am all for staging in smaller steps where ever sensible.  Something like 25% cap on extra effort to deprecate cleanly -- otherwise do the best you can for 25% extra effort?

> Revert

Not sure I follow the revert in the title, is the breaking change already merged to DFG master but its an easy undo with better staging?

Affie, 2020-07-31:
> 25% effort

I agree, in this case it should be less effort to deprecate cleanly, there's a few steps needed to change for example DynPose2. 

> Not sure I follow the revert in the title, is the breaking change already merged to DFG master but its an easy undo with better staging?

Yes, its already merged and breaks all downstream packages without a warning to track where it comes from.
The proposal will be implemented in #569 and my idea is to support both for now (At least with LightDFG). I think CGDFG can be a bit ahead for a while and only support "singleton" softtypes?

GearsAD, 2020-07-31:
Yeah, if this is breaking everything downstream, lets fix downstream first.

Affie, 2020-07-31:
So the plan is to first replace fields such as `utime` in DynPose, then types such as ContinousScalar, and then make the soft type implementations stateless.

closed 2020-07-31 by Affie

dehann, 2020-07-31:
> already merged and breaks all downstream packages without a warning

Oops, that's not great.  We definitely need to do deprecation on at least LightDFG then --thanks!