JuliaRobotics/DistributedFactorGraphs.jl #573: `smallData` as a `Dict{Symbol, Any}`

Affie, 2020-08-05:
I know this is not going to go down well, but what about: `smallData::Dict{Symbol, Any}`

It may perform better with annotations (https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-661755848) see [docs](https://docs.julialang.org/en/v1/manual/style-guide/#Avoid-elaborate-container-types-1): 

> Avoid elaborate container types
It is usually not much help to construct arrays like the following:
a = Vector{Union{Int,AbstractString,Tuple,Array}}(undef, n)
In this case Vector{Any}(undef, n) is better. 
It is also more helpful to the compiler to **annotate specific uses** (e.g. a[i]::Int) than to try to pack many alternatives into one type.

We can limit the types in the function arguments.

dehann, 2020-08-05:
The problem is how to serialize it with JSON.  By forcing a few containers the idea was to ensure JSONable?  My original suggestion for smallData was to pack it into description::String of Entry, where the value going into the `description` field is ::Dict{Symbol, String}`.  So the user is responsible for getting their data into string format.  From there we know the JSON is easy.  This also avoids the problem with `::Any`, by making the user use only string.

Affie, 2020-08-05:
Just copying this here, originally posted in https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/520#issuecomment-657323463:

- SmallData is a special data entry as you've got there, of type Dict{Symbol, Union{Int, Float64, AbstractString, Vector{Int}, Vector{Float64}, Vector{AbstractString}} or something similar
- SmallData is a special DataEntry, but it's only really used in CGDFG for saving/loading. A user won't see it, the DFGVariable will contain the Dict like above

So what I'm saying is to instead have the container as `Dict{Symbol, Any}`, but only allow the types supported by Neo4j (as above). That way it is queryable and no string parsing is needed. 

> How would that work?

something like 
`addSmallData!(dfg, label::Symbol, key::Symbol, data::Union{Int, Float64, AbstractString, Vector{Int}, Vector{Float64}, Vector{AbstractString}} )`

dehann, 2020-08-05:
ah, okay I understand now, that make sense.

dehann, 2020-08-06:
as long as the user does not work around the set and get api, because then they might use the any outside the jsonable list.

dehann, 2020-08-06:
i’d say hack it in for now and we can revisit as we start to use it more?

dehann, 2020-08-06:
also see #509

dehann, 2020-08-06:
I like this example too (thanks @Affie ): https://github.com/JuliaRobotics/IncrementalInference.jl/issues/783#issuecomment-661755848

GearsAD, 2020-08-06:
We need to add the CRUD for smalldata too, talking with Johan at the moment and we realized that doesn't exist. Need to make smalldata look like everything else.

Affie, 2020-08-06:
I don't know if it means anything, comparing SmallDataTypes to Any, Code lowered looks the same. 
llvm has a slight difference in lines. 
@code_llvm sd[:a] #Dict{Symbol, SmallDataTypes} = 89 lines
@code_llvm sd[:a] #Dict{Symbol, Any} = 79 lines

Affie, 2020-08-07:
Going with SmallDataTypes until its found to be a problem.

closed 2020-08-07 by Affie