JuliaRobotics/DistributedFactorGraphs.jl #574: Request deprecation of FileDataStore

dehann, 2020-08-06:
I'm trying to update code from about 2 months ago, and not quite sure how to proceed. 

```julia
julia> datastore = FileDataStore( joinLogPath(slam.dfg,"bigdata") )
ERROR: `FileDataStore` is deprecated, use `FolderStore`
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] FileDataStore(::String; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/Deprecated.jl:73
 [3] FileDataStore(::String) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/Deprecated.jl:73
 [4] top-level scope at REPL[35]:1
```

I see the new constructor is something like `FolderStore(key,folder)`?  Not sure what to do with key...

Affie, 2020-08-06:
Sorry, the message is not very helpful. The key was added to keep track of the stores. It still needs to be documented properly https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/563.

Affie, 2020-08-06:
The T parameter is to specify the underlying data type. The basic one is `Vector{UInt8}` for example:

`FolderStore{Vector{UInt8}}(:filestore, "/tmp/dfgFilestore")`

dehann, 2020-08-06:
Ah, thanks!  No worries, this is all still a work in progress, I'm just trying to fix other code as fast as possible.

Affie, 2020-08-06:
Perhaps we should add a default as:
`FolderStore(foldername) =  FolderStore{Vector{UInt8}}(:default_folder_store, foldername)`

dehann, 2020-08-06:
perhaps yes, that would be good

dehann, 2020-08-06:
Just for documentation, the next issue is:
```julia
┌ Warning: DFG now uses ZonedDateTime, temporary promoting and converting to DateTime local time
└ @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/Deprecated.jl:13
ERROR: MethodError: no method matching addDataEntry!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::String, ::Array{UInt8,1})
Closest candidates are:
  addDataEntry!(::AbstractDFG, ::Symbol, ::AbstractDataEntry) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/AbstractDataEntries.jl:57
```

dehann, 2020-08-06:
Were too from here
```julia
"""
$(TYPEDEF)
Abstract parent struct for big data entry.
"""
abstract type AbstractDataEntry end
```

Affie, 2020-08-06:
I may have missed some of the deprecations on `ZonedDateTime`. But I'm not following the error.

Affie, 2020-08-06:
> Also not really deprecated smoothly 

Some of it was just easier to replace completely, but I thought I kept it in deprecated.jl. 

you may be looking for just `addData!`?

dehann, 2020-08-06:
cool thanks, I'm just listing all the deprecations i encounter so we can add warnings for them (looks like new functions are there but the deprecation information is just slightly different, `addData!` here does not seem to have a `::FolderStore` option, where previous version was `::FileDataStore`).  
```julia
ERROR: MethodError: no method matching addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::String, ::Array{UInt8,1})
Closest candidates are:
  addData!(::AbstractDFG, ::Symbol, ::Union{FileDataStore, InMemoryDataStore}, ::Symbol, ::AbstractString, ::Array{UInt8,1}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/Deprecated.jl:109
  addData!(::AbstractDFG, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::Any; kwargs...) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
  addData!(::AbstractDFG, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
```

The original in this case was:
```julia
addDataEntry!( fec.slam.dfg, prevpose, fec.datastore, :JOYSTICK_CMD_VALS, "application/json", Vector{UInt8}(JSON2.write( joyVals ))  )
```

When I try replace for `:default_folder_store` I still get:
```julia
ERROR: MethodError: no method matching addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::Symbol, ::Symbol, ::String, ::Array{UInt8,1})
Closest candidates are:
  addData!(::AbstractDFG, ::Symbol, ::Union{FileDataStore, InMemoryDataStore}, ::Symbol, ::AbstractString, ::Array{UInt8,1}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/Deprecated.jl:109
  addData!(::AbstractDFG, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::Any; kwargs...) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
  addData!(::AbstractDFG, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
```

Looks like the description `::String` was lost too.  I'll go dig and add back the old command with deprecation.

dehann, 2020-08-06:
Ah, so the first problem is the deprecated function is moot since `FileDataStore` is no longer allowed:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/a6fe4ee16a12f6fcfc03d9c1786daa4fc67329f5/src/Deprecated.jl#L104

Affie, 2020-08-06:
I remember it was large changes and impossible to deprecate by pointing to new functions. So I just tried to error and give replacements. But I only covered what was in the tests.

Affie, 2020-08-06:
`description` is a keyword argument:
`function addData!(dfg::AbstractDFG, blobstore::AbstractBlobStore, label::Symbol, key::Symbol,
                  blob::Vector{UInt8}, timestamp=now(localzone()); description="", mimeType = "", id::UUID = uuid4(), hashfunction = sha256)`

dehann, 2020-08-06:
new api is like 95% match.  e.g. this function does not have a parameter for `description`:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/a6fe4ee16a12f6fcfc03d9c1786daa4fc67329f5/src/DataBlobs/services/BlobStores.jl#L73

Affie, 2020-08-06:
>  e.g. this function does not have a parameter for description:

It takes an entry. The user has to manually create the entry with the required fields such as description.

Affie, 2020-08-06:
If I remember correctly I ran into a lot of method ambiguity errors with similar methods. So I couldn't keep all of the old ones.

dehann, 2020-08-08:
Okay, so old API:
```julia
datastore = FileDataStore( joinLogPath(slam.dfg,"bigdata") )
addDataEntry!( dfg, :x1, datastore, :datalabel, "application/json", Vector{UInt8}(JSON2.write( someDict ))  )
```

new API:
```julia
datastore = FolderStore{Vector{UInt8}}(:default_folder_store, joinLogPath(fg,"bigdata"))
addBlobStore!(dfg, datastore)
addData!(fg, :default_folder_store, :x1, :datalabel, Vector{UInt8}(JSON2.write( someDict ))  )
```

However, two new problems:
- path not created then collision #578 
- compulsory MIMEType or description field #579

closed 2020-08-12 by dehann