JuliaRobotics/DistributedFactorGraphs.jl #577: Softtype as an instance or a type, T vs T()

Affie, 2020-08-08:
Softtype is now a singleton type and to avoid confusion we have to decide whether it will be used as an [instance or a type](https://docs.julialang.org/en/v1/manual/style-guide/#Avoid-confusion-about-whether-something-is-an-instance-or-a-type-1).
for example:
`getManifolds(::Type{Pose2}) vs getManifolds(::Pose2)`
For compatibility, I used an instance where @dehann used type `getManifolds(vartype::Type{LinearConditional}) = (:Euclid,)`

dehann, 2020-08-08:
I'd suggest only the Type ::T, `addVariable!(..., Pose2)`.

dehann, 2020-08-08:
its also easy to just do `T()` internally if needed, but its more difficult to go back `obj --> ::T`

dehann, 2020-08-13:
Hi Johan, how do you feel about this?
```julia
const InstanceType{T} = Union{Type{<:T},T}
```

Affie, 2020-08-17:
> Hi Johan, how do you feel about this?
> 
> ```julia
> const InstanceType{T} = Union{Type{<:T},T}
> ```

I don't know how that will help. You need something like this discouraged pattern in the style guide:
> ```julia
> foo(::Type{MyType}) = ..
> foo(::MyType) = foo(MyType)
>  ```

From a user side, 
- I think the type is easier as: `addVariable!(..., Pose2)
- I find functions of this form easier to work with: `getManifolds(::Pose2)` but I created a macro so one can generate and change it easily.

For now, my vote would be to just follow the style guide : 
>The preferred style is to use instances by default, and only add methods involving Type{MyType} later if they become necessary to solve some problem.

dehann, 2021-06-29:
> For now, my vote would be to just follow the style guide
> I think the type is easier as: addVariable!(..., Pose2)

I'm also in the camp of using the type, but I don't know how this influences type stability.  I'm assuming `::DataType` is not stable hence the suggestion to work with instances... I think we can come back to this. This feels like a minor performance, but much larger user experience thing.  Let's wait on a decision for now.

Affie, 2021-07-08:
I came across this thread https://discourse.julialang.org/t/singleton-types-vs-instances-as-type-parameters/2802/11

Maybe it can be a tie breaker.

Affie, 2025-07-08:
From triage call today, we agreed on going with instance as per the style guide internally in general. `addVariable!` will still have the convenience function that takes the type.

closed 2025-10-03 by Affie