JuliaRobotics/DistributedFactorGraphs.jl #578: Tricky Data API failures

dehann, 2020-08-08:
```julia
julia> addData!(fg, :default_folder_store, :x1, :JOYSTICK_CMD_VALS, Vector{UInt8}(JSON2.write( joyVals ))  )
ERROR: SystemError: opening file "/tmp/caesar/2020-08-08T07:50:27.654/bigdata/f348a2a9-84db-4c11-be67-cb7bcb01002f.dat": No such file or directory
Stacktrace:
 [1] systemerror(::String, ::Int32; extrainfo::Nothing) at ./error.jl:168
 [2] #systemerror#50 at ./error.jl:167 [inlined]
 [3] systemerror at ./error.jl:167 [inlined]
 [4] open(::String; read::Nothing, write::Nothing, create::Nothing, truncate::Bool, append::Nothing) at ./iostream.jl:254
 [5] open(::String, ::String) at ./iostream.jl:310
 [6] open(::DistributedFactorGraphs.var"#158#160"{Array{UInt8,1}}, ::String, ::Vararg{String,N} where N; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at ./io.jl:296
 [7] open at ./io.jl:296 [inlined]
 [8] addDataBlob!(::FolderStore{Array{UInt8,1}}, ::BlobStoreEntry, ::Array{UInt8,1}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:153
 [9] #addData!#153 at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:76 [inlined]
 [10] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime; description::String, mimeType::String, id::Base.UUID, hashfunction::typeof(SHA.sha256)) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:115
 [11] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:111
 [12] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
 [13] addData! at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99 [inlined] (repeats 2 times)
 [14] top-level scope at REPL[11]:1

julia> mkpath("/tmp/caesar/2020-08-08T07:50:27.654/bigdata")
"/tmp/caesar/2020-08-08T07:50:27.654/bigdata"

julia> addData!(fg, :default_folder_store, :x1, :JOYSTICK_CMD_VALS, Vector{UInt8}(JSON2.write( joyVals ))  )
ERROR: Data entry JOYSTICK_CMD_VALS already exists in variable
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] addDataEntry! at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/AbstractDataEntries.jl:45 [inlined]
 [3] addDataEntry!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::BlobStoreEntry) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/AbstractDataEntries.jl:57
 [4] #addData!#153 at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:75 [inlined]
 [5] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime; description::String, mimeType::String, id::Base.UUID, hashfunction::typeof(SHA.sha256)) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:115
 [6] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::FolderStore{Array{UInt8,1}}, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:111
 [7] addData!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::Symbol, ::Symbol, ::Symbol, ::Array{UInt8,1}, ::TimeZones.ZonedDateTime; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99
 [8] addData! at /home/dehann/.julia/dev/DistributedFactorGraphs/src/DataBlobs/services/BlobStores.jl:99 [inlined] (repeats 2 times)
 [9] top-level scope at REPL[13]:1
```

Affie, 2020-08-08:
The data entry was successful but the blob failed. I didn’t want to make any assumptions and delete the entry automatically. I guess we can do a quick check to see if the folder exists. 
Do we code defensively or error. The other code looks more in the error camp.

dehann, 2020-08-08:
In this case the code should create the folder because the user is asking for a specific path location.  Don't think the user should be responsible for accessing the same folder twice (feels like duplication).

Affie, 2020-08-08:
Ahh, my bad. It looks like I missed the constructor. Did it with the other store:
```julia
struct FileDataEntry <: AbstractDataEntry
    label::Symbol
    id::UUID
    folder::String
    hash::String #using bytes2hex or perhaps Vector{Uint8}?
    createdTimestamp::ZonedDateTime

    function FileDataEntry(label, id, folder, hash, timestamp)
        if !isdir(folder)
            @warn "Folder '$folder' doesn't exist - creating."
            # create new folder
            mkpath(folder)
        end
        return new(label, id, folder, hash, timestamp)
      end
end
```

dehann, 2020-08-12:
If something goes wrong in adding a a Blob after the Entry was created, then there is a collision on the duplicate Entry.  User will have to first delete the old Entry and then readd.

closed 2020-08-12 by dehann