JuliaRobotics/DistributedFactorGraphs.jl #579: Compuloray MIMEType or description in Data?????

dehann, 2020-08-08:
Old:
```julia
addDataEntry!( dfg, :x1, datastore, :datalabel, "application/json", Vector{UInt8}(JSON2.write( someDict )) 
```

New
```julia
addData!(fg, :default_folder_store, :x1, :datalabel, Vector{UInt8}(JSON2.write( someDict ))  )
```

Affie, 2020-08-08:
I don't know exactly what the issue is. Do you want it as compulsory arguments instead of keyword arguments? It makes it hard to be general.

I went on how it previously was:

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/21d238ffcf0f6d5a2f89ea10a75df320d43e4d12/src/DataBlobs/entities/AbstractDataEntries.jl#L25-L32

So keyword argument on MIME Type and added a new field `discription` also set with a keyword argument.

Affie, 2020-08-08:
Here is the function signature:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/d1f3c41714d01297a9b51fd1955a9619b1f22e9b/src/DataBlobs/services/BlobStores.jl#L107-L108

dehann, 2020-08-09:
> don't know exactly what the issue is

Okay, lets try figure that out first.  We previously spoke about always storing the format in which the data is serialized (don't think we captured that in an issue).  I think that discussion has just gotten lost and I'm trying to make sure we still capture it.  The latest example does not have a `mimetype` or a description field: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/580#issue-675548476

Previously, I did do the example with keyword: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/391#issue-604082776  but now I'm thinking the mimetype as keyword is a mistake, I think it needs to be compulsory. 

xref:
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/403#issuecomment-622978279
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/403#issuecomment-622985610

> hard to be general

Maybe I'm not understanding this part, but i think it should be a mandatory field that must be carried around everywhere (similar to e.g. timestamp).  Users **need to communicate** how the data blob was serialized otherwise its near impossible to user.  E.g., if you pull an image from the store (that someone else pushed) and it's just a binary data blob, how is a user supposed to know what they have (png, jpg, bitmap, ?).  This is more tricky if the default mimetype was assumed as `application/octet-stream`.  Hence, to me mimetype is super critical.

> v0.8 API

Yep, still figuring it out and the API has been changing quite a bit.  Currently `_uniqueKey` and default `mimetype=application/octet-stream` get locked together inside the `Entry` without the user knowing, and cannot be changed under the immutability requirement.  So the user later learns there is a mimetype and then has to redo the upload because they alse need to learn that `key=>blob` pairings are immutable.

PS, I'd almost push for a second description/note field that the user can populate.  Example, someone wants to store an acoustic wav recording so use the mimetype `application/json`, but then add the `description="this is the one with bad electronics board v0.13"`.  We also spoke about this before, but this is fine as a keyword interface that can be added after.

dehann, 2020-08-09:
This is the closest I can find about the old discussion, see the "Type" field, but we still had not quite figured it out yet: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/97#issuecomment-537984850

Affie, 2020-08-09:
I'll just restate a short explanation and history here, there are two (almost competing) implementations of storing data in the file system:
- `BlobStoreEntry` working with a `FolderStore` https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/d1f3c41714d01297a9b51fd1955a9619b1f22e9b/src/DataBlobs/entities/AbstractDataEntries.jl#L23-L32
-  `FileDataEntry` https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/d1f3c41714d01297a9b51fd1955a9619b1f22e9b/src/DataBlobs/services/FileDataEntryBlob.jl#L9-L24

I created and used `FileDataEntry`  as a first solution to quickly save data blobs. Then the stores were created with the surrounding functions, but I didn't find it flexible and it didn't fill my original need. 
I then refactored it a bit to be able to use a store in the DFG structure itself. 
That makes the `BlobStoreEntry` + `FolderStore` look like a better solution. I kept both alive (but didn't add the description and MIMEType to `FileDataEntry`.

As you can see `BlobStoreEntry` has a MIME type and a description field.

If I understand correctly, the issue then is 

- `mimeType` as a compulsory or keyword argument in the convenience functions (`addData`)
    - JT   I have never had the need for it and think it should be a bit more hidden and automated. so would say optional keyword argument (perhaps default to `application/octet-stream`, as the implementation (`blob::Vector{UInt8}`) is binary). I can see the usefulness in cases such as image/jpg etc. In those cases, it would be nice to have it automatically filled in.

- `description` as a compulsory or keyword argument in the convenience functions (`addData`)
   - JT I think its optional, so keyword argument

dehann, 2020-08-09:
thanks, added my votes above.

>  there are two (almost competing) implementations of storing data

Sounds like main effort will be a consolidation between two half solutions.  It's easier to build out new duplicate functionality and harder to update the old code.

dehann, 2020-08-11:
also see old wiki:
https://github.com/JuliaRobotics/Caesar.jl/wiki/Standard-MIME-type-for-bigData----system-wide-integration

Affie, 2020-08-11:
I like this part:
> User can upload anything,
    - Then go back and reinsert new bigData with proper MIME type declarations,

dehann, 2020-08-12:
over all design issue #594

dehann, 2020-08-12:
Related JuliaRobotics/Caesar.jl#274

closed 2020-08-12 by GearsAD