JuliaRobotics/DistributedFactorGraphs.jl #580: Add constructor to `FolderStore` to create the folder if it doesn't exist

Affie, 2020-08-08:
Like in FileDataEntry
```julia
struct FileDataEntry <: AbstractDataEntry
    label::Symbol
    id::UUID
    folder::String
    hash::String #using bytes2hex or perhaps Vector{Uint8}?
    createdTimestamp::ZonedDateTime

    function FileDataEntry(label, id, folder, hash, timestamp)
        if !isdir(folder)
            @warn "Folder '$folder' doesn't exist - creating."
            # create new folder
            mkpath(folder)
        end
        return new(label, id, folder, hash, timestamp)
      end
end
```

_Originally posted by @Affie in https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/578#issuecomment-670945504_

Affie, 2020-08-08:
Also, update test, it uses the same folder, so it was missed.

dehann, 2020-08-09:
Hi @Affie , thanks yes -- although i would rather use an `@info`.  The user is hard specifying the path and the system is responding positively to say what it has done.  Strictly speaking warnings means code needs to be changed which is not really the case here, this is desired behavior.

dehann, 2020-08-09:
The other issue with file entry is where did the MIMEtype description go? #579

dehann, 2020-08-09:
also, I lost the plot on the different between `id` vs. `key` perhaps also we can just recapture that difference here with a comment too please?

PR #584 (2020-08-09): Make directory if it doesn't exist with @info
close #580
commit 084fdc86 (2020-08-09): Make directory if it doesn't exist with @info close #580
closed 2020-08-09 by Affie