JuliaRobotics/DistributedFactorGraphs.jl #590: Serialization

Affie, 2020-08-11:
Just capturing ideas for discussion.

## History
> Another thing, we actively decided after a 2 year process not to define the "schema" for serialization. We originally had a protobuf packing approach which was hard schema type dependent and which supported forward-backward multilanguage compatibility. After much pain, we decided that top level packing MUST be JSON, and that the JSON string must at top level include the format in which the data is serialized. This is to mimic how the Internet has standardized it's data exchange format. So StructTypes.jl might be a real problem given the history on serialization.
So to stress, its probably not a good idea going down a path to unilaterally "change everything" on serialization.
In this case, it will be better to understand what the problems are with JSON.jl JSON2.jl JSON3.jl, because clearly they haven't figured it all out either. But we likely won't go down writing JSON4.jl either.
Also remember that the whole point of JSON is that serialization should be marginally or completely available outside Julia. The ZMQ interface for example is about multilanguage support and it uses the same serialization. The DB storage is definitely not a Julia implementation, later we will add a FuseDFG driver, etc, etc. A little while back we came up with the requirement that ZMQ messages should be exactly the same as FileDFG content. The schema discussion becomes a whole issue there. There are 100 different schema's available... So serialization only has 2nd best answers for everything. Whenever you pick the "best" answer for one requirement, then all others get real upset. So we have to accept a 2nd best solution for everybody and JSON is the best bet at this time.

**So JSON**

## Options
### JSON.jl + unmarshal.jl
...

### JSON2.jl 

### JSON3.jl

```julia
julia> using JSON3
julia> using DistributedFactorGraphs
julia> using TimeZones
julia> using UUIDs
julia> bse = BlobStoreEntry(:a, uuid4(), :abc, "", "", "", "", ZonedDateTime("2020-08-11T14:32:27.000+02:00"))
BlobStoreEntry(:a, UUID("a533d151-764a-425d-9e57-a8f194577866"), :abc, "", "", "", "", ZonedDateTime(2020, 8, 11, 14, 32, 27, tz"UTC+02:00"))
julia> JSON3.StructTypes.StructType(::Type{BlobStoreEntry}) = JSON3.StructTypes.Struct()
julia> json3_str = JSON3.write(bse)
"{\"label\":\"a\",\"id\":\"a533d151-764a-425d-9e57-a8f194577866\",\"blobstore\":\"abc\",\"hash\":\"\",\"origin\":\"\",\"description\":\"\",\"mimeType\":\"\",\"createdTimestamp\":\"2020-08-11T14:32:27.000+02:00\"}"
julia> JSON3.read(json3_str, BlobStoreEntry)
BlobStoreEntry(:a, UUID("a533d151-764a-425d-9e57-a8f194577866"), :abc, "", "", "", "", ZonedDateTime(2020, 8, 11, 14, 32, 27, tz"UTC+02:00"))
```

dehann, 2020-08-11:
To answer your earlier question, here is the first issue:
```julia
julia> bse = BlobStoreEntry(:a, uuid4(), :abc, "", "", "", "", ZonedDateTime("2020-08-11T14:32:27.12+02:00"))
ERROR: ArgumentError: Unable to parse string "2020-08-11T14:32:27.12+02:00" using format dateformat"yyyy-mm-ddTHH:MM:SS.ssszzz". Unable to parse date time. Expected directive DatePart(sss) at char 21
Stacktrace:
 [1] macro expansion at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Dates/src/parse.jl:104 [inlined]
 [2] tryparsenext_core(::String, ::Int64, ::Int64, ::Dates.DateFormat{Symbol("yyyy-mm-ddTHH:MM:SS.ssszzz"),Tuple{Dates.DatePart{'y'},Dates.Delim{Char,1},Dates.DatePart{'m'},Dates.Delim{Char,1},Dates.DatePart{'d'},Dates.Delim{Char,1},Dates.DatePart{'H'},Dates.Delim{Char,1},Dates.DatePart{'M'},Dates.Delim{Char,1},Dates.DatePart{'S'},Dates.Delim{Char,1},Dates.DatePart{'s'},Dates.DatePart{'z'}}}, ::Bool) at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Dates/src/parse.jl:38
 [3] macro expansion at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Dates/src/parse.jl:150 [inlined]
 [4] tryparsenext_internal at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Dates/src/parse.jl:125 [inlined]
 [5] parse at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Dates/src/parse.jl:282 [inlined]
 [6] ZonedDateTime(::String, ::Dates.DateFormat{Symbol("yyyy-mm-ddTHH:MM:SS.ssszzz"),Tuple{Dates.DatePart{'y'},Dates.Delim{Char,1},Dates.DatePart{'m'},Dates.Delim{Char,1},Dates.DatePart{'d'},Dates.Delim{Char,1},Dates.DatePart{'H'},Dates.Delim{Char,1},Dates.DatePart{'M'},Dates.Delim{Char,1},Dates.DatePart{'S'},Dates.Delim{Char,1},Dates.DatePart{'s'},Dates.DatePart{'z'}}}) at /home/dehann/.julia/packages/TimeZones/v0mfN/src/parse.jl:88
 [7] ZonedDateTime(::String) at /home/dehann/.julia/packages/TimeZones/v0mfN/src/parse.jl:87
 [8] top-level scope at REPL[8]:1
```

dehann, 2020-08-11:
From SLACK
- Just for clarification - marshaling is converting Dicts to structs and back. Serialization is technically converting the dict to a string, at least the way we talk about it
- we need to preserve the Dict <-> String behavior
- the Dict must store as a field inside itself stating what the type of that "object" is.  That was the conclusion previously

Affie, 2020-08-11:
> here is the first issue

That is a TimeZones.jl bug

dehann, 2020-09-29:
Also keep #668 in mind for future developments

dehann, 2020-11-09:
Given some new code for serializing distributions, I'm thinking we should do away with the DFG assumption that "Packed*Type" exists on the packed type name.  Instead, each variable and factor packing and unpacking converter should deal with that, and defiitely store the packed and unpacked as mimeType information inside the serialized object.  I.e. the a serialized object **MUST** have the `mimeType = application/json/PackedPose2Pose2`, or perhaps `mimeType=application/jld2/PackedPose3` etc.  That makes the provider of `PackedPose2Pose2` also responsible for having the pack and unpack `convert` functions.  The unpack `convert(<:FunctorInferenceType, ::PackedPose2Pose2)::Pose2Pose2` function is then responsible for "stripping away the `Packed`" portion of the name.  Similarly when packing.

Maybe worth doing this for both Variables and Factors, which means something about the abstraction should be perhaps `FunctorInferencedType <: InferenceType` and also `InferenceVariable <: InferenceType` (whatever the precise abstract names end up being).

dehann, 2020-12-01:
Proposal for `mimeType` field in serialized factors or variables (improvement over assuming `**Packed**MyFactor`... ):
https://github.com/JuliaRobotics/Caesar.jl/blob/e6a7fe1340a3235509275427439a38bfabcfcbc9/src/images/Pose2AprilTag4Corners.jl#L180

In other words, `mimeType = /application/JuliaLang/PackedPose2Pose2`, or `/application/json/JSONPose2Pose2`, or `/application/JuliaLang/JLD2/Pose2Pose2`, or whatever works.  Also suggest making it the first field in the packed type:
https://github.com/JuliaRobotics/Caesar.jl/blob/e6a7fe1340a3235509275427439a38bfabcfcbc9/src/images/Pose2AprilTag4Corners.jl#L153

cc @GearsAD , @Affie , @jim-hill-r

dehann, 2021-01-22:
Can use https://github.com/JuliaIO/Tar.jl for multiplatform

dehann, 2021-02-10:
Following fix to JuliaRobotics/IncrementalInference.jl#825, there is a new type stability issue when using unpack functions.  A difference in type-stability when building IIF.CCW objects can occur.  When building FMd, not all Variable type information is properly included, see:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/c6a845763448af40983755449e245d2a7e8b41dc/src/FactorGraph.jl#L632

This results in subtle differences of type.  The original dfg object has a hard type while constructing a new ccw object:
```julia
::FactorMetadata{ 
    Vector{DFGVariable{Pose3}}, 
    Vector{Symbol}, 
    Vector{Matrix{Float64}}, 
    Nothing
}
```

But at time of writing, tests in RoME:
https://github.com/JuliaRobotics/RoME.jl/blob/7fcc014d7100a1de7fa19bfd80d257fe9669574d/test/testpackingconverters.jl#L156

Results in:
```julia
::FactorMetadata{ 
    Vector{DFGVariable}, 
    Vector{Symbol}, 
    Vector{Matrix{Float64}}, 
    Nothing
}
```
notice not-concrete type `Vector{DFGVariable}`

dehann, 2021-03-17:
A good reference:
https://discourse.julialang.org/t/converting-string-to-datatype-with-meta-parse/33024/2

dehann, 2021-03-18:
Just linking `Core.eval` approach used at this point:
- https://github.com/JuliaRobotics/ApproxManifoldProducts.jl/blob/ec0661fd92ce9a824c752da4ce4d5146d7e3e4fc/src/Interface.jl#L79
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/28eec11a15ffc069a2b3f0c9481938b9de3b2eb8/src/services/Serialization.jl#L91

dehann, 2021-07-28:
Just making sure it is captured... With DFG v0.15 we changed point data to be `vnd.val::Vector{P}` where `P` is something that Manifolds.jl can understand as point representations.  How this is serialized should be reviewed and likely updated.  This all goes in concert with how serialization of new `AMP.ManifoldKernelDensity` works, e.g.:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/dd3923bea2167ca3b5b35a506102fedc389bcc39/src/SerializationMKD.jl#L47

This new serialization is still being fleshed out, so stay tuned.

dehann, 2021-08-03:
When unpacking factors and trying to get the `FactorGradientsCache!` built on the first go, the following empty `DFGVariable[]` vector design assumption causes several headaches:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/db7ff84225cc848c325e57b5fb9d0d85cb6c79b8/src/DispatchPackedConversions.jl#L46

dehann, 2022-01-31:
One of the new tricks for unordered marshaling is to use keyword constructors on the packed type:
```julia
struct MyType
  a
  b
end

MyType(;a=1, b="hi") = MyType(a,b)
```

Base already does this in automated way, along with optional values:
```julia
Base.@kwdef struct MyType
  a::Int = 1
  b::String
end

mt = MyType(b="bye")
```

Furthermore, with JSON2's use of `NamedTuple` there is an even easier way to automate all this:
```julia
# NamedTuple gets created somewhere, likely a JSON package reading text
nt = (; a=2, b="again!")

mt = MyType(;nt...)
```

See #848, and JuliaRobotics/IncrementalInference.jl#1476 for early examples.

dehann, 2022-03-19:
Alright, hows this for Marshaling with JSON?

```julia
abstract type AbstractMarshalingJSON end

Base.@kwdef struct MyType <: AbstractMarshalingDFG
  tags::Vector{Symbol} = Symbol[]
  count::Int
end

##

function marshal(::Type{T}, d::Dict{String,A}) where {T <: AbstractMarshalingDFG, A}
  _types = T.types
  _names = fieldnames(T)
  _getIdx(s::Symbol) = findfirst(x->x==s, _names)
  _dodontbroadcast(S::Type, x) = S === Any ? x : S(x)
  _dodontbroadcast(S::Type{E}, x::AbstractVector) where E <: AbstractVector = (eltype(E)).(x)
  makething(k,v) = (tmp=Symbol(k); tmp => _dodontbroadcast(_types[_getIdx(tmp)], v))
  tup = (makething(kv...) for (i,kv) in enumerate(d))
  nt = (;tup...)
  T(;nt...)
end

##

d = Dict("count" => 3)
marshal(MyType, d)

##

d2 = Dict("count" => 3, "tags" => Any["hi", "okay"])
marshal(MyType, d2)

##
```

GearsAD, 2023-01-22:
After working on CaesarInternal, I believe I have a good handle on this. My recommendation:
- Make a PackedVariable struct (same for Factor)
- Use JSON3 to encode and decode this using either an ordered struct or a unordered struct

Notes:
- JSON3 we need to omit empties, which will allow for Union{X, Nothing} optional fields.
- We need to use kwdef structs.

Advantages:
- Fast
- Marshalling looks robust and works well(ish) with GQL
Disadvantages:
- No intermediate Dict types, it's Object->String and vice versa

closed 2023-02-28 by dehann