JuliaRobotics/DistributedFactorGraphs.jl #594: <: AbstractDataEntry should implement a standard Interface

dehann, 2020-08-12:
All instances of AbstractDataEntry should at minimum implement:
```julia
label::Symbol
id::UUID
blobstore::Symbol # Perhaps this should rather be a Dict{Symbol, Symbol}?
hash::String # Probably https://docs.julialang.org/en/v1/stdlib/SHA
origin::String # E.g. user|robot|session|varlabel
description::String
mimeType::String
createdTimestamp::ZonedDateTime # of when the entry was created
```

Which will allow easy transfer between the different blob store locations and standard their use.

dehann, 2020-08-12:
xref #586

dehann, 2020-08-12:
also see:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/afd4e4154c053ff3d2e37fe965bf2e25f300a574/src/DataBlobs/entities/AbstractDataEntries.jl#L23-L50

dehann, 2020-08-12:
xref #579

Affie, 2020-08-12:
I do not agree. There should be no need for other blob store entries. `BlobStoreEntry` is the only one and should be generic enough. If it is not is should be updated.

dehann, 2020-08-12:
If only one is needed then that is even better.  I'm just looking through the code trying to fix bugs (`loadDFG!` is breaking on `BlobStoreEntry` this morning) and see there are multiple `Entry`-s...

So we can remove `MongodbDataEntry` (deprecate)?

Affie, 2020-08-12:
I left `MongodbDataEntry` there until it finds a replacement. Its the way I've been using it and for me a very convenient way to work with it. I would like to keep the InMemoryEntry though. 
I think serialization can support only the blob store entries (for now at least). It's kind of just shifting the problem to the blob stores.

dehann, 2020-08-12:
there will only be one

closed 2020-08-12 by dehann

GearsAD, 2020-08-12:
Highlander rule!