JuliaRobotics/DistributedFactorGraphs.jl #595: should AbstractDataEntry.blobstore be a ::Dict{Symbol,Symbol}?

dehann, 2020-08-12:
Since a blob is unique and can be stored in various locations, a user can retrieve it from any of those listed in `.blobstore`.  This simplifies caching persistence etc.

Related
- #594 
- #586

Affie, 2020-08-12:
How do you know where to fetch the blob from?

dehann, 2020-08-12:
Using any one of the values stored in `.blobstore`, no?

Affie, 2020-08-20:
I would say to keep the CRUD layers as simple as possible and to add complexity on top. Such as caching from different stores.  That is more or less the general design philosophy i see throughout DFG.

dehann, 2021-01-20:
Think we are (yet again) talking about different things.  What I'm talking about is blobs have a globally unique identifier and can be replicated in multiple stores at once.  Therefore, if you want to fetch the blob for a unique id, it doesn't matter which store it comes from.  This is useful for caching etc.