JuliaRobotics/DistributedFactorGraphs.jl #617: Pkg 1.5 likely breaks IIF_TEST

dehann, 2020-08-14:
IIF_TEST=true works on Julia 1.4 but not Julia 1.5:
https://travis-ci.org/github/JuliaRobotics/DistributedFactorGraphs.jl/builds/718027136

dehann, 2020-08-14:
small chance Travis testing was not showing a IIF compat issue with DFG v0.10 and instead gave the weird "UndefVarError: CloudGraphsDFG not defined" error...  Tests underway

dehann, 2020-08-14:
Special requires if we can make the tests working without CGDFG please:
```bash
SKIP_CGDFG_TESTS=true IIF_TEST=true julia -O3
```

dehann, 2020-08-14:
From SLACk:

Desire:
SKIP_CGDFG_TESTS=true IIF_TEST=true julia -O3

Don't think you can do that. You need ENV["DFG_USE_CGDFG"]="true" (edited)

dehann, 2020-08-14:
can you do the tests while skipping both IIF and CGDFG but without ENV["DFG_USE_CGDFG"]="true"?

dehann, 2020-08-14:
I think so. but runtest.jl sets it, so can't be that then:

push!(ENV,"DFG_USE_CGDFG"=>"true")
using DistributedFactorGraphs

Affie, 2020-08-14:
The main reason IIF tests exists is to test CGDFG. But I'm sure we can make it work.
Lines like this will need to be changed: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/f145c5bc0c5efcc808f49ee594b0d3396e9f0ca3/test/iifInterfaceTests.jl#L3

dehann, 2020-08-14:
okay, we can review later -- its not critical.  But then if IIF_TEST=false, then we shoudl be able to drop CGDFG requirement?

Affie, 2020-08-14:
> IIF_TEST=false, then we shoudl be able to drop CGDFG requirement?

yes, you can use  IIF_TEST=false without CGDFG, SKIP_CGDFG_TESTS=true. That's how it is with travis.

dehann, 2020-08-14:
cool thanks, will check to make sure that works -- but that will work well enough for local testing here by me.

dehann, 2020-08-14:
So thats when i get this other error:
```julia
Connectivity Test: Error During Test at /home/dehann/.julia/dev/DistributedFactorGraphs/test/interfaceTests.jl:174
  Got exception outside of a @test
  UndefVarError: CloudGraphsDFG not defined
  Stacktrace:
   [1] top-level scope at /home/dehann/.julia/dev/DistributedFactorGraphs/test/interfaceTests.jl:175
   [2] top-level scope at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Test/src/Test.jl:1115
   [3] top-level scope at /home/dehann/.julia/dev/DistributedFactorGraphs/test/interfaceTests.jl:175
   [4] include(::String) at ./client.jl:457
   [5] macro expansion at /home/dehann/.julia/dev/DistributedFactorGraphs/test/runtests.jl:52 [inlined]
   [6] macro expansion at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Test/src/Test.jl:1115 [inlined]
   [7] top-level scope at /home/dehann/.julia/dev/DistributedFactorGraphs/test/runtests.jl:50
   [8] include(::String) at ./client.jl:457
   [9] top-level scope at none:6
   [10] eval(::Module, ::Any) at ./boot.jl:331
   [11] exec_options(::Base.JLOptions) at ./client.jl:272
   [12] _start() at ./client.jl:506
```

dehann, 2020-08-14:
redoing test in #616 but with IIF upstream test branch requiring DFG v0.10

Affie, 2020-08-14:
I thought it worked. But might have broken it recently.  I’m sure its just one of the skip if dfg isa CloudGraphsDFG

dehann, 2020-08-14:
okay with new DFG master that same test https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/617#issuecomment-674265983 passed.  So something changed int he meantime, but not serious for us to find.  Thanks for the help!

dehann, 2020-08-14:
refresher, the idea of this issue is resolve IIF_TEST=true on Julia 1.5

closed 2021-02-13 by dehann