JuliaRobotics/DistributedFactorGraphs.jl #641: listSolveKeys regex filter might be failing

dehann, 2020-08-24:
MWE
```julia
using IncrementalInference
fg = generateCanonicalFG_LineStep()
[solveTree!(fg, storeOld=true for i in 1:3];

listSolveKeys(fg)
# :graphinit, :default, :default_0, :default_1, :default_2

listSolveKeys(fg, r"default_\d+") |> sortDFG
# Empty Set{Symbol}()
```

Should the regex filter not produce a list?
```julia
# [:default_0, :default_1, :default_2]
```

Affie, 2020-08-24:
> Should the regex filter not produce a list?

Doesn't look like it:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/e30c7ededb569286f746be384e67acd6f1850c69/src/services/AbstractDFG.jl#L523-L535

Affie, 2020-08-24:
Maybe try:

`filter(contains(r"default_\d+"), string.(listSolveKeys(fg)))`

dehann, 2020-08-25:
thanks, yep that is the workaround i’m using, however, the current api does accept regex but does not seem to work properly.  We should probably make this consistent with ‘ls’ regex filter pattern argument and workflow.  

We can probably just include this filter line inside the listSolveKeys function, and make sure it returns a list of ::Symbol?  I’m not sure if regex works directly on Symbol...

Affie, 2020-08-25:
It's probably better to have 2 filters:
- `filterVariables` filter as the function you wrote 
- `filterSolveKeys` additional `Symbol.(filter(contains(r"default_\d+"), string.(listSolveKeys(fg))))`

And no, regex doesn't work on Symbols

commit bc31dd90 (2020-08-28): Regex filter for solve keys in listSolveKeys close #641
PR #645 (2020-08-28): Regex filter for solve keys in listSolveKeys
close #641
commit f8716b79 (2020-08-28): Regex filter for solve keys in listSolveKeys (#645)

close #641
closed 2020-08-28 by Affie