JuliaRobotics/DistributedFactorGraphs.jl #650: Deprecated subgraph functions in CGDFG

Affie, 2020-08-31:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/eaea68c09d712532a2ad73338b0298da43e0e37a/src/CloudGraphsDFG/services/CloudGraphsDFG.jl#L465-L492

dehann, 2020-08-31:
hi, thanks for finding this. although this needs work, i would recommend deferring the effort in the short term and only return to consolidate cgdfg subgraph when it becomes really necessary or sensible to do so.  If we can limp along on current cgdfg implementation for a while, that should be okay.

Affie, 2020-08-31:
It looks like more of a maintenance thing with the functions missed with the deletion of the deprecations. 

However, CGDFG can still overwrite `buildSubgraph` and/or `copyGraph!` for performance.

dehann, 2020-09-02:
Also listing here in case it isn't/hasn't been elsewhere for DFG:
- https://juliagraphs.org/LightGraphs.jl/latest/operators/#LightGraphs.egonet-Union{Tuple{U},%20Tuple{T},%20Tuple{AbstractGraph{T},Integer,Integer,AbstractArray{U,2}}}%20where%20U%3C:Real%20where%20T%3C:Integer
- https://juliagraphs.org/LightGraphs.jl/latest/operators/#LightGraphs.induced_subgraph-Union{Tuple{U},%20Tuple{AG},%20Tuple{AG,AbstractArray{U,1}}}%20where%20U%3C:Integer%20where%20AG%3C:AbstractGraph

closed 2025-08-05 by Affie