JuliaRobotics/DistributedFactorGraphs.jl #668: Factor serialization name may need to go further

dehann, 2020-09-25:
MWE:

```julia
julia> typeof(DFG.getSolverData(f1).fnc.usrfnc!)
MixtureRelative{2,LinearRelative{1,MvNormal{Float64,PDMats.PDMat{Float64,Array{Float64,2}},Array{Float64,1}}},(:c1, :c2),Tuple{Normal{Float64},Normal{Float64}}}
```

But what currently happens is:
```julia
julia> typeof(DFG.getSolverData(f1).fnc.usrfnc!).name.name
:MixtureRelative
```

Which is not enough to reconstruct the type properly when unpacking.  Need a lot of that structure to rebuild `MixtureRelative` properly.

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/419a31d19e02b79894216ddf02a3278608ec9e80/src/services/Serialization.jl#L252

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/419a31d19e02b79894216ddf02a3278608ec9e80/src/Common.jl#L4

dehann, 2020-09-26:
HI @GearsAD , also see JuliaRobotics/IncrementalInference.jl#912 for the associated IIF code

dehann, 2020-09-26:
Just a note, all that stuff in IIF has now been merged -- so you should be able to get the latest `LInearRelative` and `Mixture_` stuff on IIF master.  Also note that future plans might consolidate `MixturePrior{..}` and `MixtureRelative{..}` into only `Mixture{..}`, but that is not critical at this time.

GearsAD, 2020-09-27:
The error above looks to be because the Packed type is [templated](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/a70c00d795ee925890ed4350fb13395abc99c93e/src/Factors/MixtureRelative.jl#L76). I think it would fail further down once that was resolved, but the short term fix it to either encode the templating into the packed type (so the packed type saves the template information and resolves the full type using it), or to have a set of concrete types for each template possibility (this, I don't think, is practical, so the first option is preferable).

dehann, 2020-09-29:
HI @GearsAD, okay thanks, let's go with the first option.  I will build it in to `PackedMixture__` then.  Thanks for taking a look!

dehann, 2020-09-29:
Ah, I learned some more.  I will fix this issue and involves two stages:
- See updated MWE for non-concert Packed types https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/668#issue-708577378
- Expand `PackedMixtureRelative` to capture all the Static Parameters contained in `MixtureRelative{N,F,S,T}` https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/668#issuecomment-699587392

closed 2020-09-29 by dehann