JuliaRobotics/DistributedFactorGraphs.jl #685: Tag v0.10.8

Affie, 2020-10-27:
@JuliaRegistrator register branch=release/v0.10

JuliaRegistrator, 2020-10-27:
Error while trying to register: "Tag with name `v0.10.7` already exists and points to a different commit"

Affie, 2020-10-27:
@JuliaRegistrator register branch=release/v0.10

JuliaRegistrator, 2020-10-27:
Registration pull request created: [JuliaRegistries/General/23715](https://github.com/JuliaRegistries/General/pull/23715)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.10.8 -m "<description of version>" 29b79aed21ea5195a9f6a51844b51e82c1dfdde7
git push origin v0.10.8
```

closed 2020-10-27 by Affie