JuliaRobotics/DistributedFactorGraphs.jl #697: retry tag v0.10.11

dehann, 2020-11-14:
@JuliaRegistrator register branch=release/v0.10

dehann, 2020-11-14:
@JuliaRegistrator register branch=release/v0.10

JuliaRegistrator, 2020-11-15:
Registration pull request created: [JuliaRegistries/General/24693](https://github.com/JuliaRegistries/General/pull/24693)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.10.11 -m "<description of version>" 5b33e8f19bcc1e76b684da65cacf8a9defe4f102
git push origin v0.10.11
```

closed 2020-11-16 by dehann