JuliaRobotics/DistributedFactorGraphs.jl #707: @dehann There is no function like that.

Affie, 2020-12-10:
From PR discussion: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/705#discussion_r540034178_

> new name is:
> 'lsfTypes(fg, LinearRelative)'

@dehann There is no function like that.
Only lsfTypes(dfg) -> Return `Vector{Symbol}` of all unique factor types in factor graph.

Did you perhaps mean just  `lsf` as the test just above that?

Should we deprecate `lsfWho` in favour of just `lsf`

dehann, 2020-12-10:
yeah that's fine, one modification: don't pass a Symbol, just pass the type (especially to resolve dispatch).
'lsf(fg, Pose2Pose2)'

This function already exists somewhere because i use it quite often.  Same goes for 
'ls(fg, Pose3)'

dehann, 2020-12-10:
@Affie , its easy to just try and see that it is already working:

```julia
using RoME

fg = generateCanonicalFG_Hexagonal(graphinit=false);

julia> lsf(fg, Pose2Pose2)
6-element Array{Symbol,1}:
 :x3x4f1
 :x4x5f1
 :x5x6f1
 :x0x1f1
 :x1x2f1
 :x2x3f1
```

Please check if there are any other loose bits around so we can close this issue as needs be.  We definitely need to close out things like `lsfWho`.

Affie, 2024-08-28:
- [ ] Deprecate `lsfWho` for just `lsf`

commit e98bc54b (2024-09-13): Deprecate lsfWho for lsf close #707
PR #1092 (2024-09-13): remove all return types and deprecate lsfWho
close #374
close #707
closed 2024-09-13 by Affie